prusa3d/PrusaSlicer #4171: Unnecessarily bulky support structure

theely, 2020-05-01:
### 2.2.0+

### MacOS 10.15.4
### Ender 3 Pro

### Behavior
Unnecessarily bulky support structure. 
Compared to other slicers (e.g. Cura) the support structure uses additional surface on the X and Y axes resulting in a very heavy structure. For the same model (see attached project) PrusaSlicer uses 30% more filament compared to Cura.

#### Cura support
[image]

#### PrusaSlicer support
[image]

#### Project File (.3MF) where problem occurs
[Hero6Mount.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4564982/Hero6Mount.3mf.zip)

agentd00nut, 2020-07-01:
I Second this;

It becomes a deal breaker when you need to print with something like TPU where supports really like to bind to the overall model.  Prusa and Slic3r itself both seem to *always* try and build from the build plate or are eager to build up additional structures when building off the actual model.

When building off the model itself cura seems to realize that it doesn't need to build up from the plate for those structures and avoids it completely.

A simple fix for Prusa might be to make a second set of support blockers and inhibitors where supports absolutely will not generate for any reason within them (or will absolutely stay within their bounds).

drelich, 2020-09-21:
Implementing this would be awesome. At this point, it does mess up things unnecessarily as viewed on the screenshot below. If the supports only generated in the 'inset' areas that are fully capable of holding the supports in full (the bottom is wider than the top of the insets) then I would not be left with unappealing scaring cause by the excessive structures.

[image]

bubnikv, 2021-10-05:
Works fine with the new "snug" supports in PrusaSlicer 2.4.0-alpha3.

[image]

closed 2021-10-05 by bubnikv

drelich, 2021-10-05:
That is indeed impressive! Just tested it on the same model I did in my earlier post:

[image]

Well done! :)