prusa3d/PrusaSlicer #4174: Reload from harddrive not available/ Name of part not loading

Fredde0869, 2020-05-02:
### Version
_2.2.0_

I use solidworks to export .3mf files and if I load these into PrusaSlicer, I cannot chose reload from harddrive and also the Name of the part does not appear. It only says bodyxxxxxxx.

neophyl, 2020-05-02:
Not really surprising. What I suspect is happening is when you load model files such as an stl into slicer and then use save project to save it as a .3mf file slicer can store the path to where the model was loaded from.  That way when you select reload from disk it can load in the current version.

When you load a .3mf file in that was created by solidworks that information as to where the parts are stored on the hard drive wont be there.  In effect you are loading in a project file that stores different information than  Prusa Slicer.  This behaviour is determined by the way Solidworks saves the file not by Prusa Slicer.

SigmaRelief, 2020-07-08:
I think Fredde0869 is talking about "importing" a .3mf, not opening the .3mf as a project.  Importing them simply adds them to the existing project so only the 3D geometry comes along.  I have this exact same issue as .3mf files do not appear to support the reload functionality which is unfortunate as is is the preferred output format, but is not fully featured as a source.

The unrelated SOLIDWORKS "feature" of assigning random names within .3mf files is also extremely anoying and makes a large platter difficult to understand, but it is hard to ask for a workaround within the slicer when SOLIDWORKS should offer more control on the file creation.  I was so frustrated with this that I wrote a python file to assign the file name to the body/bodies within the .3mf and merged it with my export macro.  [https://github.com/SigmaRelief/SOLIDWORKS-Export-to-Mesh/releases](https://github.com/SigmaRelief/SOLIDWORKS-Export-to-Mesh/releases)

I'm still adding a few features to the macro, but it is a one click way to get nice usable .3mf files (just make sure to install Python).

rtyr, 2021-12-02:
Both of these issues were fixed in PrusaSlicer 2.4.0-alpha1. Closing.

closed 2021-12-02 by rtyr