prusa3d/PrusaSlicer #4175: End_gcode Error, Why??

manurafal, 2020-05-02:
### Version: 2.0.0
### Operating system: Windows 10 64bits
### Error: 
"" G-code export to C:\Users\xxxx\AppData\Local\Temp\.1456.gcode failed due to invalid custom G-code sections:
end_gcode
Please inspect the file C:\Users\xxxx\AppData\Local\Temp\.1456.gcode.tmp for error messages enclosed between
!!!!! Failed to process the custom G-code template ...
and
!!!!! End of an error report for the custom G-code template ...""

### Description
I have the above error when I put the following code in the End_gcode:

{if filament_notes=~/.*FILAMENT_DIRTY*/}
M104 S215
{endif}

I don't know why.

bubnikv, 2020-05-03:
filament_notes is a vector variable. you should index it with [0]

manurafal, 2020-05-03:
Thanks a lot.

closed 2020-05-03 by manurafal