prusa3d/PrusaSlicer #4179: PrusaSlicer install location triggers macOS security warning about "App Translocation"

evoncken, 2020-05-02:
### Version
PrusaSlicer 2.2.0+

### Operating system type + version
macOS 10.15.4 Catalina

### 3D printer brand / version + firmware version (if known)
Prusa Mini

### Behavior
It seems the default PrusaSlicer 2.2.0+ install does not conform to "standard" macOS installation methods. It is installed as /Applications/Original Prusa Drivers/PrusaSlicer.app

This triggers a macOS security warning about "App Translocation" - the LittleSnitch firewall displays this error and refuses to add permanent rules for the app.

### Expected behavior ###
PrusaSlicer installed in Applications according to macOS "standard", avoiding security warning.

[image]

orenmazor, 2020-12-25:
I'm having the same issue

JohnOCFII, 2020-12-25:
Here's a little more about that error message from Little Snitch: 

https://help.obdev.at/littlesnitch4/alert-warnings

It would seem if PrusaSlicer was _really_ being installed improperly, more folks would be having this issue.  I'd guess there is some odd interplay happening.

orenmazor, 2020-12-25:
I agree. this is certainly not a huge issue. at least, not right now. 

my concern is that at some point, "app translocation" might become something _apple_ takes issue with, not just LS. and at that point we're gonna have a bad time.

bubnikv, 2021-01-05:
I passed this information to the guy responsible for the OSX installer.

michaelrommel, 2021-03-07:
Hi,

maybe it is worth adding, how this can be remedied in the meantime.

The dialog of Little Snitch is triggered by Apples Quarantine system:
```
❯ pwd
/Applications/Original Prusa Drivers
❯ xattr *
: com.apple.FinderInfo
: com.apple.ResourceFork
: com.apple.quarantine
New user guides Prusa3D.webloc: com.apple.FinderInfo
New user guides Prusa3D.webloc: com.apple.ResourceFork
New user guides Prusa3D.webloc: com.apple.lastuseddate#PS
New user guides Prusa3D.webloc: com.apple.quarantine
Objects: com.apple.quarantine
Pronterface.app: com.apple.FinderInfo
Pronterface.app: com.apple.quarantine
PrusaSlicer.app: com.apple.quarantine
❯ xattr -p com.apple.quarantine PrusaSlicer.app
00c3;6044cb19;Safari;2A7CAD91-F009-42B2-9DC9-DEAEAA68EAC7
❯ sudo xattr -dr com.apple.quarantine PrusaSlicer.app
❯ xattr -p com.apple.quarantine PrusaSlicer.app
xattr: PrusaSlicer.app: No such xattr: com.apple.quarantine
```

This should disable the warning by Little Snitch.

jerryau, 2021-05-05:
I was having the same issue.  Running the commands posted by @michaelrommel worked for me.

bubnikv, 2021-05-06:
From https://help.obdev.at/littlesnitch4/alert-warnings

> App Translocation is a security mechanism Apple introduced with macOS 10.12 (Sierra). If an application has not been “properly installed”, the operating system maps it to a random path before launching, usually somewhere in /private/var/folders/. This path randomization prevents loading of resources shipped alongside with the application, a mechanism often used by malware. 

> “Properly installed” means that the application must be started from a code-signed disk image or that it must have been copied to a new location in Finder.

Now I don't understand the last statement for the life of me. IMHO installing the application on OSX means to copy the application folder (PrusaSlicer.app in our case) somewhere on your hard drive. IMHO that should create a "new location in Finder". What should we do?

pickerin, 2021-09-24:
The fix to this issue is much simpler than running xattr commands.

As shipped, the Prusa Slicer comes in a folder (it's got a fancy icon, but it's a folder), inside that folder are "PrusaSlicer.app" (the application), "Pronterface" (an application that allows PrusaSlicer to send commends over USB to your printer), and an "Objects" folder.  

The "fix" is to just move all three into your /Applications folder.  It's a little messy, but Little Snitch (and macos) will be happy with it.  The permanent fix would be for both of these to be included inside of the PrusaSlicer.app application bundle, most likely in the "Resources" folder, but that will also require some re-wiring from an Application standpoint so it's looking for them there.

bubnikv, 2021-09-24:
Actually there is no more a reason to pack Pronterface with PrusaSlicer installation.
Regarding the test objects, we may separate them into their own download package.
I suppose that would solve the issue?

pickerin, 2021-09-25:
> Actually there is no more a reason to pack Pronterface with PrusaSlicer installation.
> Regarding the test objects, we may separate them into their own download package.
> I suppose that would solve the issue?

If the .app file "PrusaSlicer" file is installed in the /Applications folder, it'll solve the issue.

closed 2026-09-29 by prusaslicer-bot[bot]