prusa3d/PrusaSlicer #4180: Min and max value of variable layer height applied to wrong extruder in multi extruder mode

DavidGeorge528, 2020-05-02:
### Version
2.2.0

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Prusa MK3s with MMU2s

### Behavior
When modifying the min and max values of layer heights for different extruders in multi-material mode the edited values are applied to wrong extruder profiles when using the Variable Layer Height feature.

#### Steps needed to reproduce the problem

- Add any model to the plater
- Enable multi-material mode (e.g. Select MK3S + MMU2S profile)
- Set all extruder layer limits to the same value (e.g. Min: 0.07, Max: 0.3)
- Change extruder 2 to have the values Min: 0.15 Max 0.3
- Go to the plater and select the model, change the extruder for that model to extruder 2 and enter Variable layer height mode
- Adjust the layer height and observe the limits

#### Expected Result

The layer height limits when extruder 2 is selected is 0.15 and 0.3 and for all other extruders is 0.07 and 0.3

#### Actual Result

With extruder 2 having min and max values of 0.15 and 0.3 respectively 

[image]

The layer height for extruder 2 is still 0.07 and 0.3

[image]

However, if I select extruder 3, the limits of extruder 2 are applied

[image]

Even though the settings for extruder 3 are set to 0.07 and 0.25

[image]

The bug seems to be an off by 1 error somewhere in the code, as if I modify extruder 1 using the same values as above and change extruder 2 back to default 

[image]

The limits are applied to both extruder 1 and to extruder 2

[image]

But no other extruders

[image]

To recap, modifying the layer heights limits in the Printer settings of a multi-material extruder will apply to the next extruder in the list. E.g modifying extruder 2's limits applies to extruder 3 and so on. This is true for all extruders, with extruder 1 also applying the values to itself.

JohnnyDeer, 2020-05-04:
Hi @DavidGeorge528 , issue reproduced and confirmed, thak you for reporting.

HaythamB, 2021-11-10:
So I ran into the same problem, and from what I've found, it seems like PS creates an initial ".ini" print settings file that stores the minimum layer heights from the printer settings, which doesn't get updated when you change your printer settings min layer height values.

Under your %APP_DATA% folder for PrusaSlicer (regardless if you're using an alpha, beta, or stable release), navigate to your "print" folder, and open the ".ini" file that you're using in PS (the same filename as the profile saved under the "Print Settings" tab in PS). 

Around line 154 (for latest PS alpha at least), you'll find the setting "min_layer_height=" which is what the adaptive layer function reads from. You'll probably find that its set to the default PS value of 0.08mm, and its not reflecting the value defined under your "Printer" settings extruder. Change this min_layer_height for your extruders as needed, and that should be a fine workaround for your problem.

Prusa, please fix! ;)

vertigo235, 2022-09-13:
wow, just found this with 2.5.0 , surprising to find this issue has been open for over 2 years

vertigo235, 2022-09-13:
FWIW SuperSlicer doesn't have this issue

closed 2026-09-29 by prusaslicer-bot[bot]