prusa3d/PrusaSlicer #4183: Project name not set when opening from file.

Sembazuru, 2020-05-03:
### Version
2.2.0-win64

### Operating system type + version
Window 10 Pro
Version 1909

### Behavior
I have my filesystem set up to open .3mf files in PrusaSlicer. When I double click on a .3mf file in Windows Explorer the file loads, but PrusaSlicer hasn't set the project name to anything. When I open the same file from within PrusaSlicer the project name is set to the opened file name. This behaviour is obvious right away in the title bar of the program and by the behaviour when using the menu item File/Save.

Title bar when opening by double clicking .3mf file in Windows Explorer.
[image]

Title bar when opening same file from withing PrusaSlicer by selecting the menu item  File/Save.
[image]

#### Project File (.3MF) where problem occurs
This happens for any .3mf file that I've tried.

rtyr, 2020-05-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3751. Closing.

closed 2020-05-03 by rtyr

Sembazuru, 2020-05-03:
Sorry for the duplicate. I did try searching, but apparently I didn't use the right keywords.