prusa3d/PrusaSlicer #4184: Ubuntu Build: missing: TBB_LIBRARIES

tobyworks, 2020-05-03:
### Version
2.2.0

### Operating system type + version

Ubuntu 20 server

### 3D printer brand / version + firmware version (if known)

### Behavior

I want to build prusa slicer from source with the following:

mkdir build
cd build
cmake .. 
cmake .. -G Ninja -DSLIC3R_STATIC=1 

But im getting the following error:

-- Boost::boost exists
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  TBB library cannot be found.  Consider set TBBROOT environment variable.
  (missing: TBB_LIBRARIES) (found version "2020.1")
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindTBB.cmake:286 (find_package_handle_standard_args)
  CMakeLists.txt:323 (find_package)

It's weird because i have libtbb installed via sudo apt-get isntall libtbb-dev

Somehow i think it's not looking in the correct folder

closed 2020-05-11 by tobyworks

Megidd, 2020-06-27:
@tobyworks Did you solve the problem? How? Just curious.

spuder, 2021-12-05:
I'm seeing the same issue on OSX 11.6. I've tried the suggestions on 4959. 

```
brew update
brew install cmake git gettext
brew upgrade
git clone https://github.com/prusa3d/PrusaSlicer/
cd PrusaSlicer/deps
mkdir build
cd build
cmake ..
make
cd ../..
mkdir build
cd build
cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"
make
src/prusa-slicer
```

```
CMake Warning at CMakeLists.txt:133 (message):
  CMAKE_PREFIX_PATH element doesn't exist:
  /Users/spencerowen/Code/github/PrusaSlicer/deps/build/../deps/build/destdir/usr/local

OS X SDK Path: /Library/Developer/CommandLineTools/SDKs/MacOSX11.3.sdk
OS X Deployment Target: (default)
-- Boost::boost exists
CMake Error at /usr/local/Cellar/cmake/3.22.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  TBB library cannot be found.  Consider set TBBROOT environment variable.
  (missing: TBB_INCLUDE_DIRS TBB_LIBRARIES)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.22.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindTBB.cmake:286 (find_package_handle_standard_args)
  CMakeLists.txt:384 (find_package)
```