prusa3d/PrusaSlicer #4186: Assertion failed when slicing a file

szymonpk, 2020-05-03:
### Version

2.2.0

### Operating system type + version

Linux 5.4.35-gentoo x86_64

### 3D printer brand / version + firmware version (if known)

Not relevant

### Behavior

When I am slicing attached stl file, slicer is crashing with the following message:

```
prusa-slicer: /var/tmp/portage/media-gfx/PrusaSlicer-2.2.0/work/PrusaSlicer-version_2.2.0/src/admesh/shared.cpp:217: bool stl_validate(const stl_file*, const indexed_triangle_set&): Assertion `its.vertices.empty() || its.indices.size() == stl->stats.number_of_facets' failed.
Aborted
```

#### Project File (.3MF) where problem occurs

[strapclip.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/4571233/strapclip.3mf.gz)

mkyral, 2021-02-14:
I have the same issue in Prusaslicer 2.3 with https://www.thingiverse.com/thing:2719434 model. There are three version of excavator, but all of them causes crash when I click to Slice.

```
prusa-slicer: /var/tmp/portage/media-gfx/prusaslicer-2.3.0/work/PrusaSlicer-version_2.3.0/src/admesh/shared.cpp:217: bool stl_validate(const stl_file*, const indexed_triangle_set&): Předpoklad „its.vertices.empty() || its.indices.size() == stl->stats.number_of_facets“ nesplněn.
Aborted
```

Both eggs files slices correctly.

## System info
```
PrusaSlicer
Version:   2.3.0
Build:     PrusaSlicer-2.3.0+UNKNOWN

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.10.15-gentoo x86_64
Total RAM size [MB]: 16,694MB
OpenGL installation
GL version:   4.6.0 NVIDIA 460.39
Vendor:       NVIDIA Corporation
Renderer:     GeForce GT 740M/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
[... 284 lines cut ...]
GL_NV_texture_shader3
GL_NV_timeline_semaphore
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vdpau_interop2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
```

mkyral, 2021-02-14:
It seems like it is an issue with the model. After loading it in Blender and applying of automatic repairs, I'm able slice it without problem.

bubnikv, 2021-02-15:
I am aware of this assert. In release mode it will slice correctly I guess.

closed 2026-09-29 by prusaslicer-bot[bot]