prusa3d/PrusaSlicer #4187: Proxy secured print host URL. Special character on password

jmaeso, 2020-05-03:
### Version
2.2.0+

### Operating system type + version
MacOS 10.15.4

### 3D printer brand / version + firmware version (if known)
Custom Marlin based printer

### Behavior
Duplicating my comment on https://github.com/prusa3d/PrusaSlicer/issues/650 since looks abandoned, sorry for that.

`Hi, could be that special characters are not escaped? at least in the password part:`

[image]
[image]

`I would prefer not to change my haproxy password in the octoprint settings 😅
Am I missing something?`

TLDR:
- `http://user:password@host.com` -> works
- `http://user:password?@host.com` -> does not work

bubnikv, 2020-12-03:
Fixed with 1b720dbc3d8411c28285844866e89ba3a84884bb
Indeed, a raw URL was passed to libcurl, while it should have been URL encoded.
Thanks for the report.
The fix will be in PrusaSlicer 2.3.0-beta2.
Closing.

closed 2020-12-03 by bubnikv

bubnikv, 2020-12-03:
The commit 1b720db was not correct, I had to revert it.
I was calling the escape function over the complete http:/xxx/yyy line, which is incorrect. Certainly the http:// prefix must not be escaped.

I believe we have either to leave the escaping of special characters to you 
https://www.werockyourweb.com/url-escape-characters/

or we have to add new input fields for user / password, escape them and combine them into an URL. That is a lot more work though and I don't think that the user base is large enough to justify the effort.