prusa3d/PrusaSlicer #4189: Extrusion multiplier is applied after Max volumentric speed

yschroeder, 2020-05-03:
### Version
2.2.0

### Operating system type + version
WIndows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2.5S Firmware 3.9.0-RC3

### Behavior
PrusaSlicer applies the `Extrusion multiplier` after `Max volumetric speed` of the filament. This leads to situations where `Max volumetric speed` is not the hard upper limit on extruded filament per second.

I use the `Max volumentric speed` to print at high speeds close the maximum melting capability of the hotend. It my case it is configured to `15 mm3/s`, anything higher and the extruder starts losing steps. My print takes 20 minutes in this case and print settings are configured to constantly extrude at `15 mm3/s`, verified by the volumetric speed preview. If I now set the extrusion multiplier to anything higher than `1.0`, the extruder starts losing steps.

You can easily reproduce this by setting the extrusion multiplier to e.g. `2`. The print should take twice as long, as you extrude double the amount of material and volumetric speed is at its limit already. However, PrusaSlicer gives the same estimated print time.

I think PrusaSlicer should apply the `Max volumetric speed` after the `Extrusion multiplier`. In the example above, the print time should be doubled when setting the `Extrusion multiplier` to `2`.

TeaFlavoredTea, 2024-12-03:
This is causing me some trouble too. I'm trying to dial in some 85a TPU on the CHT nozzle with a mk4S on PS 2.8.1. Since the soft TPU jamming is the limiting factor and not printing speeds, I expected to be able to use Max volumetric speed like a throttle. Its unintuitive that once I have the Max volumetric speed dialed in to avoid jamming, if I want to correct for under/over extrusion I'll need to modify both Max volumetric speed and Extrusion multiplier.

It seems that the correct procedure is to first tune the under/over extrusion with Extrusion multiplier and then push Max volumetric speed up.

yschroeder, 2026-09-03:
This ticket is still valid.

Reproduction:

* Set all extrusion rates to travel speed (300mm/s)
* Set `Max volumentric speed` to 10mm3/s
* Slice object
* Check volumentric flow rate preview: Complete object is printing at 10mm3/s (correct)
* Change extrusion multiplier in filament to 2
* Slice object
* Check volumentric flow rate preview: Complete object is printing at 20mm3/s (wrong)

Expected outcome:
Changing the extrusion multiplier to 2 will still print at 10mm3/s as this is the maximum volumentric flow the machine is capable doing.

Actual outcome:
Extrusion multiplier is applied in the wrong order changing the actual volumentric flow to 20mm3/s and overloading the extruder.