prusa3d/PrusaSlicer #4190: Sanity Checks are broken

yschroeder, 2020-05-03:
### Version
2.2.0

### Operating system type + version
Windows 10

### Behavior
There are multiple issues with the sanity checks that are done by PrusaSlicer when changing print settings.

First, sanity checks should not block me from slicing an object, at least when I am in expert mode. They can give a warning, that hints at the possible problem, but it should not be a hard error.

I will now go through the sanity checks and their problems:

#### Layer height can't be greater than nozzle diameter
This might not be reccomended, but it should not be a hard error. I have a print that works perfectly fine when doing this. I am using a 0.4mm nozzle with 0.5mm layer height and 1.9mm extrusion width. I am basically printing a 2D part consisting of 2 layers and only perimeters. I tricked PrusaSlicer into allowing me to do this by changing the nozzle size in the printer tab to 1.0mm (while using a 0.4mm nozzle in reality).

#### Excessive extrusion_width=1.9 mm to be printable with a nozzle diameter 0.4 mm
Again, this works perfectly fine in my application. Still, PrusaSlicer considers this a hard error.

#### Parameter validation: extrusion_width (Do you mean 1.1% instead 1.1 mm?)
This one is just very annoying. It pops up every time you change one of the extrusion widths. While the sanity check itself seems fine to me, why would it start to warn at 1.1 mm? Who would want to print at 1.1% extrusion width of the layer height anyway? It might make sense to start to warn about this at let's say 20mm/20%.

#### top_infill_extrusion_width=0.4 mm is too low to be printable at a layer height 0.4 mm
You get this message even when you disabled top infill completely (number of layers set to 0). The field to change the top_infill_extrusion_width is grayed out when you set the top infill to 0. Still, this is a hard error and you cannot print. You have to set the number of top layers >0, then you can change the top infill width to something that does not trigger the sanity check, then you can set top infill to 0 again. The same problem exists for the other infill types as well.

yschroeder, 2022-01-04:
A slight bump to this issue. They were noted again in tis video (starting at 4:54):
https://www.youtube.com/watch?v=tJHnx89Fmi0

bubnikv, 2022-01-04:
> Parameter validation: extrusion_width (Do you mean 1.1% instead 1.1 mm?)This
one reports now above 50 in PrusaSlicer 2.4.0

yschroeder, 2026-09-03:
This ticket is still valid.

All of the below reproductions were done on a `Prusa MK4S` printer on preset `0.20mm SPEED` in Prusa Slicer v3.0.0-alpha11.

### (First) Layer height can't be greater than nozzle diameter

Reproduction:

* Raise layer height above 0.4mm at 0.4mm nozzle diameter
* Error message prevents slicing
* Multiple other error messages pop up that indicate extrusion width is too small for current layer height

Expected outcome:
* Warning message warns about unusual settings. Can be ignored by clicking `I know what I'm doing`.

### Extrusion width is too large to be printable with the nozzle diameter

Reproduction:

* Raise extrusion width above 1.2mm at 0.4mm layer height
* Error message prevents slicing

Expected outcome:
* Warning message warns about unusual settings. Can be ignored by clicking `I know what I'm doing`.

### Extrusion width is too low to be printable at current layer height

Reproduction:

* Set layer height to 0.4mm
* Error message prevents slicing
* Set top solid layers to 0
* Error message still prevents slicing

Expected outcome:
* Warning message warns about unusual settings. Can be ignored by clicking `I know what I'm doing`.

## Workaround

The most curious part is that the slicer does not use the nozzle diameter for anything else but these sanity checks (AFAIK). So a viable workaround is to lie to the slicer about the nozzle diameter. You can then set the required settings as needed.

## Why is this relevant?

There are prints that are only 2D perimeters with a height. For example a tube standing upright. In these cases it is easily possible to print such an object even with a 0.4mm nozzle at high extrusion widths and layer heights. This results in fast sturdy prints with a glass-like transparency.

Personally, I found these problems when trying to optimize print settings for this object:
https://www.printables.com/model/30406-surgical-mask-strap-light

This video gives some insights: https://www.youtube.com/watch?v=tJHnx89Fmi0