prusa3d/PrusaSlicer #4191: Unable to use GUI elements on a laptop screen, they are off the side

DeviousPenguin, 2020-05-03:
### Version
Version: 2.2.0+
Build: PrusaSlicer-2.2.0+-202003211132

### Operating system type + version
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Catalina Version 10.15.4 (Build 19E287)
Total RAM size [MB]: 17,180MB

### 3D printer brand / version + firmware version (if known)
Longer LK4 Pro running Klipper Firmware

### Behavior
Preview Mode has GUI checkbox elements: 'Travel', 'Retractions', etc. along the bottom, but some are are off the edge of my screen, and unable to use them all, here's a screenshot:

[image]

I'm not really sure what the ideal fix for this is, I'm not aware of any screen resolution requirement to run PS. If the 'export G-Code' button on the right hand side was a little smaller, would this solve the issue? I'm not really sure how many checkboxes I'm missing!

#### Project File (.3MF) where problem occurs
This happens with each and every file

Thanks

FidelCapo, 2020-05-04:
Thank you for report. You are right. This can be problem on small screens. I think, this will be improved in next version.

DeviousPenguin, 2020-05-04:
That's great, thank you :)

kenthinson, 2020-05-22:
@ DeviousPenguin what mac do you have? what screen resolution? is the app off the top of the screen? cant tell as the image has everything cropped out (dock, menus, ect)

DeviousPenguin, 2020-05-22:
@kenthinson Hi it's a MacBook Pro (13-inch, Mid 2012), and the screen resolution is 1280 x 800. Here's a full screen shot with dock, menus, etc. :)

[image]

kenthinson, 2020-05-22:
@DeviousPenguin Ah I get it now! Sorry when you said off the edge of the screen I was a bit confused. The check box hiding under the right hand panel.  Yes that’s something that will need a little work to fully fix.

DeviousPenguin, 2020-05-22:
No problem, I should have included a full screenshot when creating the issue. Thanks for looking into a fix for this.

closed 2026-09-29 by prusaslicer-bot[bot]