prusa3d/PrusaSlicer #4192: Huge wipe tower when supports are enabled

Tillema, 2020-05-04:
### Version
PrusaSlicer 2.2.0

### Operating system type + version
Windows 10 Home

### 3D printer brand / version + firmware version (if known)
Custom printer (doesn't matter, issue occurs in the slicer and is not configuration-dependent)

### Behavior
* When I slice a multi-material (for dual-nozzle printer) while supports and the wipe tower are enabled, it generates a extravagant huge wipe block that never ever will fit the bed.
* 1) Load a multi-material object
  2) Go to print settings and change:
      - Enable Wipe Tower
      - Turn on support (any kind)
  3) Go to printer settings and change:
      - Number of extruders to 2.
  4) Slice and look what happens.
* Expected result: just a small wipe block that looks 'normal' instead of this huge.
* Actual result: Unprintable huge wipe block
* Screenshot:
[image]

According to me, this issue is new. Delete if that's not the case.

#### Project File (.3MF) where problem occurs
[Heftruck.zip](https://github.com/prusa3d/PrusaSlicer/files/4573933/Heftruck.zip)

Hope you can help, otherwise I have to manually edit the G-code, but that is a bit too much time consuming

lukasmatena, 2020-05-04:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3919. The suggested workaround solved the issue when I tried. This is really a bug in 2.2.0 release.

Tillema, 2020-05-04:
Oh well, searched for a half hour, but couldn't find any similar issue. Sometimes the titles are a bit meaningless, so, I didn't notice it when scrolling through the list of issues. Thanks you for your feedback. I'll check whether it works on previous versions.

Tillema, 2020-05-04:
I already see what it causes. The _layer height_ is bigger than the _maximum layer height_. I'm not an expert in programming, but in my opinion there are two solutions:
- Ignoring the _maximum layer height_ set in printer settings (this is now only used by the adaptive layer height);
- Decrease the _layer height_ and ignoring the set layer height in print settings if the _maximum layer height_ is lower than the _layer height_ (there also need a '<' changed to a '<=', because when the _layer height_ is equal to the _maximum layer heigth_, this issue still occur).

EnriqueSubias, 2020-06-01:
I ran to the same problem ([reddit post](https://www.reddit.com/r/prusa3d/comments/gu7adn/huge_wipe_tower_when_enabling_supports_in/)), thanks for the solution!!!, I tried everything I could until I found this post.

lukasmatena, 2020-06-03:
This should be fixed by ed98a859af7c591a7d13aee9d0a0391c258738e2.
Thanks for the report. Closing.

closed 2020-06-03 by lukasmatena