prusa3d/PrusaSlicer #4193: Time to slice

DavidCPreston, 2020-05-04:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 64 bit latest build running on fairly new HP Ryzen laptop with 8GB Ram and 256GB SSD

### 3D printer brand / version + firmware version (if known)
Geeetech A10 Marlin 1.1.8

### Behavior
Load STL
Slice
Wait 20 minutes before export to GCODE. 
An STL of about half the size takes about half the time.
Cura takes 90 seconds.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs

[N3DPS v7 8 stack.zip](https://github.com/prusa3d/PrusaSlicer/files/4576499/N3DPS.v7.8.stack.zip)

rtyr, 2020-05-04:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/2213, https://github.com/prusa3d/PrusaSlicer/issues/2899. Disabling "Avoid crossing perimeters" will help. Unfortunately the "avoid crossing perimeters" feature can slown down slicing dramatically in some cases.

DavidCPreston, 2020-05-04:
Thank you so much

Regards

David Preston

closed 2020-05-04 by rtyr