prusa3d/PrusaSlicer #4195: Marking safe areas for sequential printing

pedjas, 2020-05-05:
### Version
2.2.0

### Operating system type + version
Not relevant

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
When sequential printig is turned on PrusaSlicer warns user if objects are to close for extruder to pass them safely. That is good.

But there is no visal indication what is safe distance around objects. User has to guess. this leads to non optimal space usage.

This a new feature request.

It would be good to have option to display safe area around each object. That would help user see where exactly objects may be located for optimal space usage and overall printing speed.

#### Project File (.3MF) where problem occurs
Irellevant

foreachthing, 2020-05-05:
No, _arrage_ should work with the correct offset. That would make much more sense.

pedjas, 2020-05-26:
Sure, arrange is expected to set correct offset. But, if user arranges objects by himself interface lacks of indication what is safe distance.

ulab, 2021-03-01:
This should also be shown for the wipe tower in case you are using the "no sparse layers" feature.

rtyr, 2021-08-12:
Implemented. It will be included in the upcoming PrusaSlicer 2.4.0-alpha1. Closing.

[image]
[image]

closed 2021-08-12 by rtyr