prusa3d/PrusaSlicer #4196: Option to disable solid infill

nife87, 2020-05-05:
PrusaSlicer v2.2.0+win64

I have had trouble finding out if this feature is already or will be on the roadmap, so I will ask here. Please close the issue and redirect me if the request is already in progress.

If I am ever to use PrusaSlicer again (just tried it for the first time and generally liked it) there needs to be an option to disable the solid infill entirely (the current limiting options are not enough).
For instance, on the screenshot of the CHEP cube I have attached, the light blue and purple areas near the perimeters are printed with solid infill.
In this case, solid infill makes the print duration at best more than twice as long (59m vs. 27m. in Cura) as when using a slicer without the solid infill feature (original Slic3r, Cura etc.).
I understand why the feature could contribute to better prints but it is certainly unnecessary for some prints, costing both unnecessary time and filament.

[image]

jesusbackflips, 2020-05-10:
Ahhhhh you can set Infill from 20% to 0%. Please Close :)

nife87, 2020-05-10:
No, because I still need the "regular" infill. There is a huge difference between the cubic infill and the solid infill.

jesusbackflips, 2020-05-10:
What you mean? What you want?

You wanna reduce "red" Infill?
blue Bridgesupport (cant disable this one)?
or get rid of Pink stuff? aka "ensure vertical shell Thickness"

nife87, 2020-05-10:
Like I described in the original post, I would like to remove the light blue and purple areas, which seems to be called solid infill in PrusaSlicer.

For the sake of reference, this is an image of the same cube in Cura with approximately the same settings and view. You will notice that there are only two kinds of "lines": Shells/walls (blue, corresponding to the yellow lines in PrusaSlicer) and infill lines (green, corresponding to the red lines in PrusaSlicer). This kind of infill is what I want to achieve in PrusaSlicer.
[image]

jesusbackflips, 2020-05-10:
Aaaaah now :) 
Check this Picture i hope i catched all Problems ;) 
[image]

nife87, 2020-05-10:
Thank you for the response.

I actually already previously tried both enabling and disabling that "vertical shell thickness" option without any luck, at least from what I could determine visually (well, I think tried nearly all options on every tab one-by-one, because I assumed I missed something somewhere).

So, naturally, I tried to disable that option again after your response and repositioned the cube like suggested (first on its side and then on its bottom to realign it).

Unfortunately, it did not help in the slightest (newly sliced from same perspective and layer):
[image]

Used print settings (I copied the settings from your image, also just in case):
[image]

As I am unaccustomed to PrusaSlicer, please let me know if exporting my entire settings (using Export -> Export Config?) would help if I am supposedly at fault here.

I am unable to upload STL files directly, so I will link to the specific STL file in use:
https://cdn.thingiverse.com/assets/b9/7c/5d/a3/c9/CHEPCalibrationCube.stl

jesusbackflips, 2020-05-10:
Yeeeah^^ You scale Up the Cube right?. How the printer is able to Print in Air`??

nife87, 2020-05-10:
What do you mean? I do not scale anything (at least not on purpose). The cube is 20x20x20mm in the STL, which is also its size in the slicer (100%).
[image]

nife87, 2020-05-10:
But now that you mention scale, the perimeters/walls actually seem awfully thin compared to Cura, even when defined as 3 perimeters.

That is strange because the extrusion width is defined as 100% (0.5mm) for the perimeters:
[image]

nife87, 2020-05-10:
Ah, finally. Just tried the default 0.2 normal settings again (though I tried that many times before - I know how to debug erroneous settings), and now I suddenly get a different image with normal-sized walls and without the solid infill:
[image]

So apparently something went terribly wrong with my previous settings (though I do not know precisely what- the scale and the rest of the settings seemed correct), causing the walls to become very thin (and requiring the solid infill), which would also agree with the statement that you thought I scaled up the cube. Had I simply noticed that the walls were much thinner than in Cura, I could have saved myself a lot of trouble, so I learned as much.

Thanks for the help. I will take it from here and close the issue.

closed 2020-05-10 by nife87