prusa3d/PrusaSlicer #4198: Choppy framerate with G-Sync/FreeSync enabled monitors on Windows

cnf, 2020-05-05:
### Version
2.2.0

### Operating system type + version
Windows 10, build 19619.1000

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Laggy with G-sync enabled.

With a G-Sync (compatible) monitor, and g-sync enabled, frames drop immensely when using prusaslicer, to a point it is very unpleasant.

Other programs have the same issue, such as Windows Terminal, with the relevant bug here: https://github.com/microsoft/terminal/issues/649

It seems that when using GPU rendering, and not sending enough updates, the g-sync triggers, and drops the framerate to as low as it can.

The _workaround_ is to add the program (in this case PrusaSlicer) to the nvidia control panel exception list, and disable g-sync for that program.

The solution on PrusaSlicer end seems to be to update the screen more often when using GPU rendering.

JimBim89, 2023-09-17:
Thought my display had issues, but it was PrusaSlicer constantly swapping between 40 and 240 fps.

This causes flickering on my oled and can only be avoided by whitlisting PrusaSlicer to not use G-Sync for me.

IBNobody, 2023-11-27:
https://github.com/microsoft/terminal/issues/649#issuecomment-491288003

Those are the exact instructions to fix this.

closed 2026-09-29 by prusaslicer-bot[bot]