prusa3d/PrusaSlicer #4199: Bottom interface layer is not generated (long open defect)

kursatu, 2020-05-05:
### Version
2.2.0
This has been around for few years , as far as I can remember.
Can this be finally fixed?
It greatly diminishes the value of multi-material printing.
Here is my model which has the missing bottom interfaces marked with red.
The top interface layer is generated and printed fine.
Here is the thread in the foeum.
https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-general-discussion-announcements-and-releases/bottom-interface-layer-is-not-generated-long-open-defect/#post-212771
[image]
[image]

bubnikv, 2021-02-24:
Please provide a 3MF to test with.

bubnikv, 2021-03-18:
No data provided, however I suppose the issue will be solved by trimming the support columns with the object silhouette. In that regard it is a duplicate of #915 #1340

bubnikv, 2021-03-19:
Fixed with 9f09f03228478aa77549efec845717b06e46cbc3

closed 2021-03-19 by bubnikv

bubnikv, 2021-03-22:
Fixed with 9f09f03