prusa3d/PrusaSlicer #4200: Cannot build deps since wxWidgets changes

nebbywan, 2020-05-05:
### Version
Commit 5980d46e4ce3f48289789f81be38f7ed292852ad

### Operating system type + version
Win10 Pro x64 v2004 build 19041.208
MSVS 2019 v16.5.4

### Behavior
I'm guessing since commit 572b5ba8bbf7903d4c0b67f6f441002ee1680cab, I have been unable to build the deps, with two machines and the CLI methods failing on the following errors:

[image]

2bd524849ad8aeeb8fb96f29c9ef9d28dfeaa727 builds fine.

tamasmeszaros, 2020-05-06:
Hello! I can't reproduce your issue, the dependencies build just fine with the version of MSVC you described. Did you compile the dependencies on a fresh cmake cache?

nebbywan, 2020-05-06:
@tamasmeszaros Unfortunately, yes. Even a clean install of MSVS and Win10 on one machine with a reclone of the repo in a separate location.

tamasmeszaros, 2020-05-07:
@nebbywan I'm sorry, I still  can't reproduce your issue. It could help if you'd post the whole build output as a log file.

tamasmeszaros, 2020-05-07:
I would maybe revisit the file system paths. Do you use spaces in your directories or some other obscurities? Maybe an overly lengthy path?

nebbywan, 2020-05-07:
Here's the outputs from CMake and Build; sorry I didn't post them before.
[cmake output.txt](https://github.com/prusa3d/PrusaSlicer/files/4594097/cmake.output.txt)
[build output.txt](https://github.com/prusa3d/PrusaSlicer/files/4594098/build.output.txt)

I don't think it would be the path (it's "C:\Users\zacha\Downloads\PrusaSlicer\deps") since there's nothing really odd about it and I'm able to successfully build the deps from earlier commits. Please let me know if there's anything else that might be helpful or that you'd like me to try.

tamasmeszaros, 2020-05-08:
Ok, I'll take a look at them. Don't worry we'll find the issue ;)

tamasmeszaros, 2020-05-12:
Hi @nebbywan ! So, could it be that you have the include files from an older wxWidgets in `C:\Users\zacha\Downloads\destdir-64\usr\local\include\wx` ? The build output states that and it should not be the case. The includes should be from the checked out source code of wx. Try to delete the destdir-64 folder content and restart the build.

nebbywan, 2020-05-12:
@tamasmeszaros Well, that was it! Is this an intended behavior? I just find it a bit odd that it only started doing this recently and that it would look in there for anything other than putting newly built files. Thanks for the help!

tamasmeszaros, 2020-05-13:
@nebbywan, I'm glad it works. 

I found many times that with CMake, life is never boring. I will look into it as its not an intended behavior for sure, but I guess we have to be ready for these.

wx was not updated in our deps for a long time so you could experience no issues before. I recommend that you do a compilation with a completely clean build directory, and be skeptic with CMake in general.

closed 2020-05-13 by tamasmeszaros

nebbywan, 2020-05-13:
@tamasmeszaros Understood. Once again thanks for the help and all your work on this project. Stay safe friend!