prusa3d/PrusaSlicer #4201: Modifier splits shell

gnydick, 2020-05-06:
### Version
2.2.0+win64

### Operating system type + version
Windows10 64bit Version 1909 (OS Build 18363.778)

### Behavior
When I add a modifier and slice, the sliced preview paths are split where the modifier intersects the model as if the modifier intersection was a separate shell from the main model.

[shell split by modifier.zip](https://github.com/prusa3d/PrusaSlicer/files/4584753/shell.split.by.modifier.zip)

rtyr, 2020-05-06:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/3317, https://github.com/prusa3d/PrusaSlicer/issues/3742, https://github.com/prusa3d/PrusaSlicer/issues/3846.

Modifier volume with different settings will split the object into pieces, which are sliced independently. This is the way how PS work with modifiers at the moment, so it is expected behavior.

bubnikv, 2021-06-03:
Duplicate, closing this one.

closed 2021-06-03 by bubnikv