prusa3d/PrusaSlicer #4204: inconsistent layers

rhinosan, 2020-05-06:
### Version
Version: 2.1.0+
Build: PrusaSlicer-2.1.0+-201909160907

### Operating system type + version
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Version 10.14.6 (Build 18G4032)
--- also happens on Windows

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 Pro - Stock Model

### Behavior
* inconsistent layers in some models. does not happen with all models (cannot be filament or printer issue). if same print repeated, same layers are affected every time. if any parameter changed, other layers may be affected. 

*
; generated by PrusaSlicer 2.1.0+win64 on 2020-05-05 at 06:27:43 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M73 P0 R255
M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P2000 R1500 T2000 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M104 S235 ; set extruder temp
M140 S100 ; set bed temp
G1 Z20 ; this is a good start heating position
G28; Home XY
M84 ; disable motors
M190 S100 ; wait for bed temp
M109 S235 ; wait for extruder temp

G1 X20.0 F7200;
G1 Y12.0 F7200;
G1 Z0.3 F7200;

M117 Purge extruder
G92 E0 ; reset extruder

G1 X200.0 Y12.0 Z0.3 F1500.0 E15 ; draw 1st line
M73 P0 R255
G1 X200.0 Y12.4 Z0.3 F5000.0 ; move to side a little
G1 X20.0 Y12.4 Z0.3 F1500.0 E30 ; draw 2nd line
G92 E0 ; reset extruder
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
M117 burn da cats
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M900 K200; Filament gcode
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.15

#### Project File (.3MF) where problem occurs
[image]
[image]

thanks, ideas appreciate!

rtyr, 2020-05-06:
Provide your 3MF project file please.

rhinosan, 2020-05-06:
[simple_test_unconsist_layers.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4587731/simple_test_unconsist_layers.3mf.zip)

rtyr, 2020-05-08:
It seems similar to https://github.com/prusa3d/PrusaSlicer/issues/332. 
I tried to print it and I saw some issues as well with the cubic pattern. It was completely ok with honeycomb and grid.

rhinosan, 2020-05-11:
thanks, seems to work so far!!

bubnikv, 2020-11-20:
PrusaSlicer 2.3.0-alpha4 controls anchoring of the infill lines to the innermost perimeter better:

The anchoring lines never overlap
The open ends of infill lines are anchored as well.
I believe it should improve this particular print.

Please retest with the upcoming PrusaSlicer 2.3.0-alpha4. Closing.

closed 2020-11-20 by bubnikv