prusa3d/PrusaSlicer #4206: Z Banding aligned to holes in vertical wall

Makhaira, 2020-05-06:
### Version
2.2.0

### Operating system type + version
W10

### 3D printer brand / version + firmware version (if known)
Mk3, Bondtech BMG extruder, 3.9.0 RC3 firmware

### Behavior
Printing an object with holes in the vertical wall results in a Z band around the entire perimeter of the object, coinciding with the bottom and top layer of the hole.

G-code preview gives no indication of change in extrusion width, speed or cooling.

HOWEVER

If I import the same STL, and my config from PS into Slicer++, the part prints without issue.
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
[Banding_Problem.zip](https://github.com/prusa3d/PrusaSlicer/files/4588577/Banding_Problem.zip)

rtyr, 2020-05-07:
There are more factors involved here in my opinion, see some issues with label "gap fill". Slicer++ produced better result here simply because it sliced it hollow inside, without any gap fill or internal perimeters (your 3MF settings).

Also perimeters at Z5.2-7.2 are printed in different order, so there might be different internal stresses and cooling effects of the filament, which can lead to different appearance (and this is ABS). 

Try this configuration, it should help: 
extrusion width for perimeters/external perimeters 0.4 -> **0.45**
gap fill speed 0 (disabled) -> **40**
detect thin walls **ON**

I tried to print your gcode with the same Verbatim ABS filament and I can see similar issue, altough it is barely visible in my case (it took me some time to actually make it visible on the photo). It was OK with suggested changes.

[image]

[image]

Makhaira, 2020-05-07:
Thanks for taking the time to look into this, I really appreciate it.
Whilst I can clearly see you fixed it, I don't fully understand why. Could
you help with a few questions please?

1) why change the extrusion width to. 45?
2) Why does adding gap fill help? Or are you purely turning that on to
avoid a gap for strength reasons?
3) why are the perimeters in a different rider and how could you tell that?
Gcode viewer?

Many thanks,going to try printing with your settings tonight.

rtyr, 2020-05-08:
> 
> 1) why change the extrusion width to. 45? 
>2) Why does adding gap fill help? Or are you purely turning that on to avoid a gap for strength reasons? 

To achieve single gapfill line instead of two internal perimeters on this particular model, it was not meant as universal setting.

>3) why are the perimeters in a different rider and how could you tell that? Gcode viewer? 

Because of the holes. Yes, I previewed generated gcode.

Makhaira, 2020-05-08:
>    1. why are the perimeters in a different rider and how could you tell

Great thanks. Which viewer to you prefer? I use ncviewer but find it hard
to follow.

JBollow, 2022-09-19:
I encountered  that problem myself and tried a lot. For me it came down to disable Z-Hop after retract in Cura.

rongith, 2023-04-04:
Can you try new arachne perimeter generator for that issue? And tell if it solves this?

closed 2023-09-12 by rtyr