prusa3d/PrusaSlicer #4207: Gcode set temperature higher than filament setting

tphan26363, 2020-05-06:
### Version
_Version of PrusaSlicer 2.2.0 64 bit
- Operating system Window 10 Pro 64 bits
_Version 10.0.18362
-Custom Printer Dual extruder .  
-Firmware: 
[Extruder-R.V5.zip](https://github.com/prusa3d/PrusaSlicer/files/4588867/Extruder-R.V5.zip)
[Extruder-R.V5.Gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/4588874/Extruder-R.V5.Gcode.zip)

Marlin 2.0.0

### Behavior
The G code generated by Prusa slicer contain setting below:
Line 6430: M104 S235 T0 ; set temperature
* _Steps needed to reproduce the problem_
 Load the model in to  Grusa slicer generate G code
* _Expected Results: Max temperature setting for Extruder 1 should be 230
* _Actual Results: There is a line in Gcode setting temperature to 235

rtyr, 2020-05-07:
It is caused by your "Ooze prevention" settings., see attached screenshot. Closing.

[image]

closed 2020-05-07 by rtyr

tphan26363, 2020-05-07:
Great catch!!  I missed that.