prusa3d/PrusaSlicer #4209: 2.3.0alpha - assert issue in ToolOrdering::fill_wipe_tower_partitions

rongith, 2020-05-07:
### Version
2.3.0alpha

### Operating system type + version
Windows

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
I had to comment out 
`assert(! m_layer_tools.empty() && m_layer_tools.front().has_wipe_tower);`
to make the program running in debug mode under visual studio 2019. I think then the condition in the assert may not be good. As I'm not sure to understand what it does exactly, I didn't do any modification about it.

For a simple part and without MMU, I have:
m_layer_tools.empty() = false
and 
m_layer_tools.front().has_wipe_tower = false
and then assert condition is false.

rongith, 2020-09-04:
Corrected by 8ecb0bfe

closed 2020-09-04 by rongith

lukasmatena, 2020-09-04:
Thanks. I somehow missed this issue. I'm not sure if this assert is critical now, but the commit I reverted caused more issues than just this.

rongith, 2020-09-16:
I won't report anymore minor issues like that which doesn't affect the release version. It is not valuable enough for your workflow. It is already hard to bring down the issues number.

lukasmatena, 2020-09-16:
@rongith No, use your judgement and report whatever you feel might be an issue (just not immediately the day the code was committed, in that case it may be a work in progress). We are thankful if people help use find bugs before the release.
Failing assert is an issue from my point of view. Someone put it there for a reason and might not be aware of a use case when it fails. There is nothing wrong about reporting that - not to mention that the amount of people running debug builds is quite low, so it is quite likely to be missed by a regular user.