prusa3d/PrusaSlicer #4210: building

Tinchus2009, 2020-05-07:
### Version
Latest source files

### Operating system type + version
win10 64b

Hello. Im trying to build from sources, Im follwng the steps from https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How to build - Windows.md

I installaed visual studio 2019 as described in the noob section. I compiled the dependencies with the NO DEBUG option.

But when running cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="C:\PrusaSlicer-master\usr\local"  I get this:

C:\PrusaSlicer-master\build>cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="C:\PrusaSlicer-master\usr\local"
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
-- The C compiler identification is MSVC 19.25.28614.0
-- The CXX compiler identification is MSVC 19.25.28614.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: C:\PrusaSlicer-master\usr\local (from cache or command line)
Building without Win10 Netfabb STL fixing service support
-- Found Boost 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET REQUIRED COMPONENTS system;filesystem;thread;log;locale;regex;chrono;atomic;date_time
-- Found boost_headers 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/boost_headers-1.70.0
-- Found boost_system 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/boost_system-1.70.0
--   libboost_system-vc142-mt-x64-1_70.lib
-- Adding boost_system dependencies: headers
-- Found boost_filesystem 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/boost_filesystem-1.70.0
--   libboost_filesystem-vc142-mt-x64-1_70.lib
-- Adding boost_filesystem dependencies: headers
-- Found boost_thread 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/boost_thread-1.70.0
--   libboost_thread-vc142-mt-x64-1_70.lib
-- Adding boost_thread dependencies: chrono;headers
-- Found boost_chrono 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/boost_chrono-1.70.0
--   libboost_chrono-vc142-mt-x64-1_70.lib
-- Adding boost_chrono dependencies: headers
-- Found boost_log 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/boost_log-1.70.0
--   libboost_log-vc142-mt-x64-1_70.lib
-- Adding boost_log dependencies: atomic;chrono;date_time;filesystem;regex;thread;headers
-- Found boost_atomic 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/boost_atomic-1.70.0
--   libboost_atomic-vc142-mt-x64-1_70.lib
-- Adding boost_atomic dependencies: headers
-- Found boost_date_time 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/boost_date_time-1.70.0
--   libboost_date_time-vc142-mt-x64-1_70.lib
-- Adding boost_date_time dependencies: headers
-- Found boost_regex 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/boost_regex-1.70.0
--   libboost_regex-vc142-mt-x64-1_70.lib
-- Adding boost_regex dependencies: headers
-- Found boost_locale 1.70.0 at C:/PrusaSlicer-master/usr/local/lib/cmake/boost_locale-1.70.0
--   libboost_locale-vc142-mt-x64-1_70.lib
-- Adding boost_locale dependencies: system;headers
-- Found Boost: C:/PrusaSlicer-master/usr/local/lib/cmake/Boost-1.70.0/BoostConfig.cmake (found version "1.70.0") found components: system filesystem thread log locale regex chrono atomic date_time
-- Boost::boost exists
-- Found TBB: C:/PrusaSlicer-master/usr/local/include (found version "2017.0")
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = C:/PrusaSlicer-master/usr/local/include
--   TBB_DEFINITIONS         = __TBB_NO_IMPLICIT_LINKAGE
--   TBB_LIBRARIES           = C:/PrusaSlicer-master/usr/local/lib/tbb_static.lib
--   TBB_DEFINITIONS_DEBUG   = TBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     =
--   TBB_DEFINITIONS_RELEASE =
--   TBB_LIBRARIES_RELEASE   = C:/PrusaSlicer-master/usr/local/lib/tbb_static.lib
-- Found CURL: C:/PrusaSlicer-master/usr/local/lib/libcurl_a.lib (found version "7.58.0")
-- Found EXPAT: C:/PrusaSlicer-master/usr/local/lib/expat.lib (found version "2.2.0")
-- Found ZLIB: C:/PrusaSlicer-master/usr/local/lib/zlib.lib (found version "1.2.11")
-- Found PNG: C:/PrusaSlicer-master/usr/local/lib/libpng16_static.lib (found version "1.6.35")
-- Found OpenGL: opengl32
-- FindGLEW: did not find GLEW CMake config file. Searching for libraries.
-- FindGLEW: GLEW_USE_STATIC_LIBS: ON.
-- FindGLEW: GLEW_INCLUDE_DIR: C:/PrusaSlicer-master/usr/local/include
-- FindGLEW: GLEW_INCLUDE_DIRS: C:/PrusaSlicer-master/usr/local/include
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for SHARED: .lib
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for STATIC: .lib
-- FindGLEW: GLEW_SHARED_LIBRARY_RELEASE: C:/PrusaSlicer-master/usr/local/lib/glew.lib
-- FindGLEW: GLEW_STATIC_LIBRARY_RELEASE: C:/PrusaSlicer-master/usr/local/lib/glew.lib
-- FindGLEW: GLEW_SHARED_LIBRARY_DEBUG: GLEW_SHARED_LIBRARY_DEBUG-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY_DEBUG: GLEW_STATIC_LIBRARY_DEBUG-NOTFOUND
-- FindGLEW: GLEW_SHARED_LIBRARY: C:/PrusaSlicer-master/usr/local/lib/glew.lib
-- FindGLEW: GLEW_STATIC_LIBRARY: C:/PrusaSlicer-master/usr/local/lib/glew.lib
-- FindGLEW: GLEW_LIBRARIES: C:/PrusaSlicer-master/usr/local/lib/glew.lib
-- FindGLEW: GLEW_VERSION_MAJOR: 1
-- FindGLEW: GLEW_VERSION_MINOR: 13
-- FindGLEW: GLEW_VERSION_MICRO: 0
-- FindGLEW: GLEW_VERSION: 1.13.0
-- Found GLEW: C:/PrusaSlicer-master/usr/local/include (found version "1.13.0")
-- FindGLEW: Creating GLEW::glew_s imported target.
-- FindGLEW: Creating GLEW::GLEW imported target.
-- Found NLopt in 'C:/PrusaSlicer-master/usr/local/lib'.
-- Using NLopt include directory 'C:/PrusaSlicer-master/usr/local/include'.
-- Using NLopt library 'C:/PrusaSlicer-master/usr/local/lib/nlopt.lib'.
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find OpenVDB (missing: OpenVDB_LIB_COMPONENTS openvdb) (found
  suitable version "6.2.1", minimum required is "5.0")
Call Stack (most recent call first):
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindOpenVDB.cmake:280 (find_package_handle_standard_args)
  CMakeLists.txt:444 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/PrusaSlicer-master/build/CMakeFiles/CMakeOutput.log".
See also "C:/PrusaSlicer-master/build/CMakeFiles/CMakeError.log".

In CMakeError.log there is this :

Determining if the include file pthread.h exists failed with the following output:
Change Dir: C:/PrusaSlicer-master/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe cmTC_8fc7c.vcxproj /p:Configuration=Debug /p:Platform=x64 /p:VisualStudioVersion=16.0 /v:m && Microsoft (R) Build Engine version 16.5.0+d4cbfca49 for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) C/C++ Optimizing Compiler Version 19.25.28614 for x64

  CheckIncludeFile.c

  Copyright (C) Microsoft Corporation.  All rights reserved.

  cl /c /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"cmTC_8fc7c.dir\Debug\\" /Fd"cmTC_8fc7c.dir\Debug\vc142.pdb" /Gd /TC /errorReport:queue "C:\PrusaSlicer-master\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c"

C:\PrusaSlicer-master\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c(1,10): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PrusaSlicer-master\build\CMakeFiles\CMakeTmp\cmTC_8fc7c.vcxproj]

Why are this files missing, 

Can someone help me to do the build?
Thanks in advance

kijai, 2020-05-07:
Getting the exact same if I try building dependencies without debug.
`Could NOT find OpenVDB (missing: OpenVDB_LIB_COMPONENTS openvdb) (found
suitable version "6.2.1", minimum required is "5.0")`

tamasmeszaros, 2020-05-07:
Please try the latest master

tamasmeszaros, 2020-05-07:
You should also remove the Debug configuration form the list `CMAKE_CONFIGURATION_TYPES` if you intend to not use it. Otherwise, in a multi config generator, the debug libraries will be required on windows msvc.

kijai, 2020-05-07:
> 
> 
> Please try the latest master

Got it working with the those changes, thanks!

tamasmeszaros, 2020-05-07:
I'm glad it works, thank you for the report.

closed 2020-05-07 by tamasmeszaros