prusa3d/PrusaSlicer #4212: Export gcode de and Remove SD button

Davidrenaud314, 2020-05-07:
### Version 2.2.0

### Win 10 pro 64 bit
### I3MK3S latest version
### Behavior
additional buttons to save directly on SD card are great.
[image]

But the “eject button” appear on the screen before the end of the Gcode check and if I click eject to early, I get a message:
[image]

 
I need to wait the end of the export process before eject. In that case I get no error.
I may recommend that eject button appear after the export process is completed.
 

* _Steps needed to reproduce the problem_
Export a Gcode on SD card and click eject as soon as it appear to get error message. Wait end of the process to click and message do not appear.

JohnnyDeer, 2020-05-12:
Hi @Davidrenaud314 , issue reproduced and confirmed, thank you for reporting.

rtyr, 2021-02-17:
Fixed. Closing.

closed 2021-02-17 by rtyr