prusa3d/PrusaSlicer #4214: Wrong default values shown in the bubble help

tkapler, 2020-05-07:
### Version
2.2.0+ Windows 10/64 bit

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
Prusa mini - latest/not important

### Behavior
The default values shown in the popup bubbles above the input fields are different from the real default values.
E.g. go to Print settings / Skirt and brim:
Distance from object: the real default value is 2 mm, the default value shown in the bubble: 6 mm
Skirt height: real default value is 3, the default value shown in the bubble: 1
...

Idea - put theese default values from the bubbles out, when it is problem to update this information. On the left of each input is ⬤ symbol, when the value equals to the default one (maybe better would be e.g. ✔ symbol), and anticlockwise symbol ↺ when there is a different value then default and clicking this will revert it to default - so here the program "knows" what should be the default value so the popup there could say something like "revert to default value: 1"

rtyr, 2020-05-08:
Those are application default values, not Prusa3d system profiles default values. If you create a new custom printer profile using configuration wizard you will see, that real values are corresponding to "bubble" values. Closing.

closed 2020-05-08 by rtyr

tkapler, 2020-05-11:
@rtyr sorry, i am "just" an user, but IMO your resolution of this issue is wrong. I (as an user) do not care where do the defaults come from. The point is, that it says, that the default value is X, but the real default value is Y and even when i click to revert to default value, it revert to Y not X. Therefore that useless (in this context) X value should not be shown at all. What is the purpose of such misleading and irrelevant information here?
If those values are default for setting up system profiles, than such defaults could be during the setting up the profiles only.

barrenaedu, 2020-09-27:
I agree with @tkapler it is very confusing to use the app as you do not know if there is a bug or something wrong. As a prusa user I opened the app,  I selected the default prusa profile, and when I go to a setting (ie Steam Position) I see the profile uses Nereast but the hint/help says "Default value: Align" (?) Does that have any common sense? mmm