prusa3d/PrusaSlicer #4216: Y-axis stops working at a specific height at one STL only

tehdusk, 2020-05-08:
### Version
2.2.0+win64

### Operating system type + version
Win 10 x64 v1909

### 3D printer brand / version + firmware version (if known)
Creality Ender 5, Marlin 1.9
Silent board

### Behavior
Y-axis stops working above ~29mm height at one particular STL file. Works fine with other STLs.

Tried twice with the same gcode, could replicate it's not a mechanical issue (like a loose connector).
Tried a third time after slicing the model again. Printed several other things successfully in the meantime, also higher than the "critical" height. Problematic STL failed again in the same way with new gcode though.

Here are some pictures of the failed prints. 
[image]
I think it's obvious where the Y-axis stopped working every time.
[image]

#### Project File (.3MF) where problem occurs
Preview:
[image]

Project:
[PROJECT_MaskV2silicone_Mask-WithFilterSupport.zip](https://github.com/prusa3d/PrusaSlicer/files/4598495/PROJECT_MaskV2silicone_Mask-WithFilterSupport.zip)

"Unprintable" STL:
[STL_MaskV2silicone_Mask-WithFilterSupport.zip](https://github.com/prusa3d/PrusaSlicer/files/4598496/STL_MaskV2silicone_Mask-WithFilterSupport.zip)

Generated GCode. Old code for tries 1+2, newer code for try 3.
[GCODE_Mold+for+a+Silicone+Seal.zip](https://github.com/prusa3d/PrusaSlicer/files/4598512/GCODE_Mold%2Bfor%2Ba%2BSilicone%2BSeal.zip)

tehdusk, 2020-05-08:
So unfortunately I just reproduced the error with another STL. Same height, same problem. See picture. 

[image]

This only happened with PrusaSlicer. I verified this by printing older GCode made with Slic3r, which prints fine above 29mm height.

rtyr, 2020-05-08:
This is clearly an issue with your printer. The gcode is fine, you can check it yourself in text editor. Gcode commands for Y movement are there, your printer just not doing them. Maybe some issue with stepper driver.

To prevent any doubts, I printed both your gcodes without issues. Closing.

closed 2020-05-08 by rtyr