prusa3d/PrusaSlicer #4217: Feature request: multiple print host options

ruralaccent, 2020-05-08:
Version 2.2.0
Win10
MK3S

I use the Printer Host function regularly to upload sliced files to my Octoprint server. However, I have three separate Octoprint servers on three different Pis, and thus I have to either change this information manually or store three different printer profiles with different host data (which I have to maintain as otherwise identical). This is not efficient.

Is it possible to add multiple print host profiles, rather than just have one single one? This would enable multiple servers to be stored and selected via drop-down or something? It would be much more efficient and user friendly, not least as printer profile isn't really a comfortable home for print host data, as the latter is somewhat separate.

[image]

Many thanks!

ayourk, 2020-05-10:
I am also interested in seeing this feature come to the light of day.  I have only 1 printer but 2 different ways to access this printer.  One of the ways is via OctoPrint, the other is via FlashAir.

I have some ideas on implementation of this feature.

[image]
You can have the G> button open up a drop down with a menu listing available host types, and from each host type, list available host aliases under there.  Some sort of "Add Host" button would be needed to add more than 1 host and some sort of unique naming alias system would be needed so as to identify each sub host.  Example XML follows (includes OP's scenario too):

`<printerhosts>
  <host>
    <type>OctoPrint</type>
    <alias>Prusa Ferdinand</alias>
    <address>192.168.5.2:5000</address>
    <password>AEE30B658....</password>
    <certificate></certificate>
  </host>
  <host>
    <type>OctoPrint</type>
    <alias>Ender 3 Daisy</alias>
    <address>192.168.5.7:5000</address>
    <password>BF6325EA....</password>
    <certificate></certificate>
  </host>
  <host>
    <type>FlashAir</type>
    <alias>Prusa Butterfly</alias>
    <address>192.168.5.16</address>
    <password></password>
    <certificate></certificate>
  </host>
</printerhosts>`

And the example menu would look something like:
OctoPrint >
FlashAir >

When you would hover over OctoPrint, the 2 OctoPrint hosts' aliases appear.  When you hover FlashAir, the 1 FlashAir host alias would appear.  Host types that don't have any aliases are NOT shown in the drop down menu.  Clicking on the designated host sends the sliced data to that host.  If there is ONLY 1 print host, no need to use the aliases or the drop down menu.  For simplicity in coding, the single print host could have an alias of "Default".  In addition to being able to add hosts, of course a way to edit and delete hosts would be needed.

bubnikv, 2020-10-17:
We have implemented it the other way around. One physical printer profile allows an assignment of multiple logical printers.
It will be part of PrusaSlicer 2.3.0-alpha1.
Closing.

closed 2020-10-17 by bubnikv