prusa3d/PrusaSlicer #4218: Excessive material at perimeter end

kookaburra88, 2020-05-09:
### Version
PrusaSlicer 2.2.0+win64

### Operating system type + version
Win 64

### 3D printer brand / version + firmware version (if known)
Anycubic Predator + Marlin 

### Behavior
I noticed excessive material at the end of outline perimeters of 20mm calibration cube (and other objects), see photo.
[image]

Filament diameter, extrudion multiplier & lin_advance were properly calibrated. Investigation of the Gcode reveals, that the start and end point of the outer perimeter are (nearly) equal. 
Here a snippet of the Gcode generated via PrusSlicer:
G1 X9.800 Y9.800 F10800.000
G1 F1200.000
G1 X-9.800 Y9.800 E0.54121
G1 X-9.800 Y-9.800 E0.54121
G1 X9.800 Y-9.800 E0.54121
G1 X9.800 Y9.740 E0.53956.

The start point is at coordinates (9.8, 9.8)mm while the end point is at (9.8, 9.74)mm. The linewidth and nozzle width are both set to 0.4mm. According to my understanding, this forces the last extruded line to overlap with the first extruded line. 
My expectation for the last move would be 
G1 X9.800 Y9.60 E0.53569, thus printer stops extruding (linewidth/2 = ) 0.2 mm before the center of the first line start point. Depending on the flow math, perhaps even 0.4mm before the center of the first line start point as the extruded lines do not only have an extension in with but also in length.

[xyzCalibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/4602700/xyzCalibration_cube.zip)

riri19, 2020-05-10:
Same for me :(

jesusbackflips, 2020-05-10:
Try out to adjust "jerk" Settings. (In Slicer, Printersetting) default around 10
maybe 8 or 6. if lower Value slows down in sharp angles

kookaburra88, 2020-05-15:
> Try out to adjust "jerk" Settings. (In Slicer, Printersetting) default around 10
> maybe 8 or 6 (or 12- max. ~16) . if lower Value slows down in sharp angles.
> Higher causing more Vibration and harder Turn
> EDIT: It can oozing at Edge when Jerk + Speed + yada causing near a Start/Stop motion

Already tried with no success. Furthermore, lin_adv was developed to compensate corner bulging.

closed 2026-09-29 by prusaslicer-bot[bot]