prusa3d/PrusaSlicer #4219: Excessive Travel with "avoid crossing perimeters"

nikscha, 2020-05-09:
### Version
2.2.0

### Operating system type + version
win 10 64 bit
### 3D printer brand / version + firmware version (if known)
custom duet based delta

### Behavior
enabeling "avoid crossing perimiters" option leads to excessive travel

happens every time I activate the option

expected result: activating the option should result in travel moves that don't cross perimiters.

actual result:
there is an EXCESSIVE amount of travel moves after I turn the option on.
The image with less moves has the option turned off.

[image]
[image]

#### Project File (.3MF) where problem occurs

[Rugged_Organizer_Top_V2.zip](https://github.com/prusa3d/PrusaSlicer/files/4603280/Rugged_Organizer_Top_V2.zip)

jesusbackflips, 2020-05-10:
Like "Travelmove out the Box" :)

bubnikv, 2020-12-11:
[image]

"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.