prusa3d/PrusaSlicer #4221: Hole through stl disappears after slice

dodohand, 2020-05-10:
### Version
Version 2.2.0+linux-x64

### Operating system type + version
Linux Mint 19.1 Tessa, up-to-date kernel 4.15.0-99-generic

Seems to be rendering fine.

### 3D printer brand / version + firmware version (if known)
Issue in Slicer, Will eventually print on a Prusa MK3S, 0.4mm nozzle, but still backordered

### Behavior
- STL seems to import correctly
- Configured to 0.2mm layer hight - same result in QUALITY and SPEED settings
- Slice
- One hole through part (out of 8 "identical" holes) goes missing in slice ( x~=2cm, y~=15cm)
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[hex_key.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/4604952/hex_key.3mf.gz)

rtyr, 2020-05-10:
I was unable to reproduce it on 2.2.0-win64. Reparing the model will most likely fix the issue. Right click on the model and use "Fix throught the Netfabb" option. The model is actually marked as broken in win64 PrusaSlicer.

[image]

dodohand, 2020-05-10:
Interesting. When I reopen from the .3mf the model slices correctly. Here's the source .stl :
[hex_key.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4605879/hex_key.stl.zip)

Netfabb is a Windows only program, so I don't have access to it.

When I right-click on the model it shows a checkmark by "Printable", and shows as "Manifold: Yes" in the "Info" dialog area, so I'm not sure that the .stl imports with an error in Linux.

Using the 'File->Repair STL file...' to create hex_key.obj and importing the .obj instead of the .stl resluts in the same initial sliced output - that one hole missing:
[image]

My Linux system info, in a bit more detail:
PrusaSlicer
Version:   2.2.0+linux-x64
Build:     PrusaSlicer-2.2.0+linux-x64-202003211856

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 4.15.0-99-generic x86_64
Total RAM size [MB]: 16,727MB
OpenGL installation
GL version:   4.6.0 NVIDIA 440.64.00
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1080/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
[... 318 lines cut ...]
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vdpau_interop2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum

jesusbackflips, 2020-05-10:
Check "Print resolution Value" in Printsettings
Check "Wall width"
Check "Layer Heigth"
Check "# Contur
Try around with this Settings.

rtyr, 2020-05-10:
@dodohand I can confirm, that I was able to reproduce your issue with provided STL file. It was OK after I repaired it using online Netfabb service (not application). This feature is unfortunately not available in Linux PrusaSlicer, I wasn't aware of that.

rtyr, 2021-06-09:
It's sliced correctly on current master. Likely fixed with https://github.com/prusa3d/PrusaSlicer/commit/9c5dd358ab23e9351dce0b9c71eaa5ddf92d6f17 (STL is created in OpenSCAD). Closing.

closed 2021-06-09 by rtyr