prusa3d/PrusaSlicer #4227: Feature request: alternative "bridge infill" pattern for better prints

dewi-ny-je, 2020-05-11:
### Version
Prusa slicer 2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Printrbot LC + Klipper firmware

### Behavior

I am in the process of tuning my printer for better bridging.

During the process, and in several videos online showing the same, I can see the same behaviour when the bridging settings are not yet optimal, but almost there: the overhang perimeters (the first and last group of bridge lines) become already perfect, but the bridge infill (the inner ones) still sag or disconnect altogether. At the end, when the settings get better, both the outer and inner stay flat.

Let me show Prusa Slicer and a print:

[image]

[image]

This happens because the overhang perimeters have time to cool down (they are laid down one side and then the other, alternatively), while the bridge infill is done sequentially one after another.

As result, the bridge infill after being extruded gets heated again by the nozzle while extruding the next string. Let's not forget that the metal surface of the nozzle is usually at least 3x as big as the orifice, meaning that the previous bridge infill string gets "ironed" again by the metal surface of the nozzle.

Instead of laying down the bridge infill one after another, **changing the order of the bridge infill to skip one or two strings and then do those on a second** (and if needed a third) **pass**(es) would allow the bridge infill strings to **cool down naturally** much more, resulting in the following:

- much simpler tuning of bridge settings
- less need for forced cooling with a fan, meaning that materials which are sensitive to drafts (ABS, nylon, PC, ...) will print better because the intensity of the air movement is lower.

Obviously there is a price to be paid, which is a non-continuous extrusion, since in the second (and eventually third) passes the head will cross its previous path, and in that (extremely short) period of time, no filament has to be extruded. I think that this is a very minor price to be paid, but it may be relevant for Bowden setups. I suggest the "alternate bridge infill pattern" to be user-selectable and not a default one.

dewi-ny-je, 2020-05-11:
Another example of how much tolerance in the settings it could be possible if natural cooling is further exploited: in the following image the overhang perimeters are basically always holding well, even if the print speed was changed from 15 to 50 mm/s. The rest... not so much.

[image]

HeartOfGermany, 2020-10-08:
You have quite a thick diameter, or am I wrong? With that you really have to print extremely slow with high fan speed. Might also consider printing with a thinner layer (adaptive slicing!). This cools them much quicker and offers quite an improovement on the quality.

Yet I do not like the bridging and overhang quality of any slicer. From hand written code I can tell, that a 90° overhang (perfectly horizontal printing) actually is possible with quite okay surface (not even optimized) and perfect angular accuracy. However slicers are not that smart. It is to complicated for me to dig into the code and assembly, because it requires quite a lot of know how to get it work properly. Otherwhise I would have tried to implement this.

neophyl, 2020-10-08:
You can’t print with a thinner layer on bridges in PS. All bridge extrusion is done at the nozzle width so layers are 0.4 for the standard nozzle. As the bridge layer has nothing to squish against all calculations assume that the extrusion matches nozzle. 
Bridging is one of the areas most lacking when compared to some other slicers on comparison unfortunately. Still the best for most other things though.

RichiH, 2020-10-09:
Another use case for this is more artistic bridge infill on skeleton prints. As you can see, the default infill is both ugly and slower here.

[image]

RichiH, 2020-10-09:
Update: It's particularly horrible from below :p

[image]

Patsjemoe, 2021-07-31:
Hi,
I Just started with prusa.
Did you solve this ? 
I am wondering if an infill at 90 degree could solve this. This way, distance is shorter.
Regards

dewi-ny-je, 2021-07-31:
No the developers didn't implement it yet.

For info for the developers: I think Superslicer does it and it's a fork of Prusaslicer, so it may be easy to port it.

Patsjemoe, 2021-07-31:
Hi,
Is 90 deg fill  in enhanced settings not a solution ?
Regards

dewi-ny-je, 2021-07-31:
We are not talking about normal infill, we are talking about bridging. The bridge infill has different rules and you cannot pick the direction.

Patsjemoe, 2021-08-01:
Hi,
you can , in Print settings, Infill Advanced Bridging angle in Expert mode.
(see attachment)
regards
Ludo

dewi-ny-je, 2021-08-01:
> Hi, you can , in Print settings, Infill Advanced Bridging angle in Expert mode. (see attachment) regards Ludo
> […](#)

Ok I didn't know about it and I should try it sometimes, but my request of using alternative bridge infill lines may apply to this 90° case as well: the idea is that by laying down bridge infill lines spaced apart (1 every 1, then doing the missing ones in a second pass) the nozzle will not be as close and will not heat as much the one which as just been laid down (and which is still hot), resulting in 1) better bridge infill everywhere, in every situation 2) much easiert tuning of bridge settings.
Using 90° (if it works, which I'm not sure about) it's a workaround.

Actually, if you use 0° you can see from my first image above that the bridge infill is anchored very well to the previous layer, there is quite some overlap. If I use 90° the only anchoring is the thin surface in contact with overhang perimeters:

[image]

I doubt it will work well and it is not solving the original issue, but I can try.

Patsjemoe, 2021-08-02:
Hi,
Hope you share the result....
Regards

wrongbad, 2022-10-29:
I'm facing a similar challenge to @RichiH and I believe an "easy" improvement for us would be a drop-down to choose "bridge infill pattern" and allow "concentric". 

This may help with the rectangular bridge case as well since the spiraling inward would alternate sides on each pass rather than always neighboring the last strand.

dewi-ny-je, 2022-10-30:
My solution is currently to use Superslicer which allows setting a percentage of bridge infill, so that by setting 50% one every two lines is skipped and there is a gap to allow air to pass and cool better.
Similar to what I asked.

closed 2026-09-29 by prusaslicer-bot[bot]