prusa3d/PrusaSlicer #4229: PrusaSlicer-2.2.0 does not import body name if filetype is 3mf

hackbender, 2020-05-12:
### Version
_Version of PrusaSlicer used goes here_
2.2.0+win64
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Pro 10.0.17763 build 17763
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Pre-assembled Prusa i3 MK3S
### Behavior
* _Describe the problem_
Files imported do not retain file name when importing 3mf files.  Not an issue on earlier versions or .stl files.
* _Steps needed to reproduce the problem_
Import file that is a 3mf type.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
File exported retains part name without being manually renamed.
* _Actual Results_
Each 3mf import needs to be renamed
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
Both parts are identical except for filetype.  Parts drawn and exported using Solidworks Premium 2019 SP5.0
_Is this a new feature request?_
No

rtyr, 2020-05-16:
Can you provide some 3MF file which causes this issue?

hackbender, 2020-05-16:
> 
> 
> Can you provide some 3MF file which causes this issue?

Yes, here you go.
[sm funnel.zip](https://github.com/prusa3d/PrusaSlicer/files/4638694/sm.funnel.zip)

rtyr, 2020-05-17:
Thanks, I was able to reproduce it, but it seems that PrusaSlicer is just using the name included in the 3MF file.

[image]

hackbender, 2020-05-17:
It seems the issue is with Solidworks.  I will close this issue.  Thanks for looking into it.

closed 2020-05-17 by hackbender