prusa3d/PrusaSlicer #4230: [2.2.0] Spiral Vase option not working in combination with modifiers

th33xitus, 2020-05-12:
### Version
2.2.0

### Operating system type + version
Win10

### Behavior
When creating a Shape-Cylinder which should be printed with Spiral Vase mode and trying to modify it with modifier geometries, slicing the model gives the following error:
```
The Spiral Vase option can only be used when printing single material objects
```
There is no material change configured and the printer is configured with a single extruder.
Therefore the object is actually a single material object.

JohnnyDeer, 2020-05-13:
Hi @th33xitus , reason for this error is that modifiers separate model to single parts. That mean, where is used modifier, there is "another" model and slicer classify it as multi part print.

fiveangle, 2020-06-25:
While that may be the internal design, the limitation doesn't make sense to the user.

Perhaps the simplified approach of using the existing routines within PS that is used with "Export plate as STL..." could be used to temporarily process-then-consolidate the model behind the scenes into a single part then passed back to the existing spiral vase process ?  This would not resolve the problem of trying to add a "hole" modifier to the solid base layers (as I found was impossible and led me here) but it should resolve the otherwise incompatibility of current vase mode with other geometry modifiers. It may sound like a bit of a hack, but I don't see why it wouldn't work.  That's how I manually resolved the problem myself.

bubnikv, 2021-06-03:
PrusaSlicer 2.4.0-alpha1 will allow "negative volume" aka "subtractor" volumes to be added to a spiral vase object.

We need to make a decision about the modifiers.

Snuff1eupagus, 2021-12-07:
You know what would really be nice? to have the ability to add spiral Vase as a height range modifier, to a non-spiral vase print.
That would be a powerful tool.
For now the application of negative volumes has given me something to work with.

Thank you

JonBerlin, 2022-01-13:
Is there an update to this issue? Is this going to be addressed in an upcoming version?

DanaFo, 2022-06-15:
Ugh I just hit this too trying to do a max flow calibration on a vase object with multiple layer heights. C'monnnn.

TilmanGuenther, 2025-06-20:
This is an extremely annoying bug. I would like to have a spiral vase, and add variation to extrusion width as the height changes, but currently my workaround is manual G-Code edits which is an absolute pain. Please fix!

Snuff1eupagus, 2025-06-21:
@TilmanGuenther 
I would like to assure you that you can add any modifier, to spiral vase mode, however you will not find any member of the prusa development team, that can or will show you how to do this, it's been years! that this has been possible. no one will show it here, that knows because of how prusa currently handles the software development, showing this here would be to the communities detriment. we would lose the ability rather then helpfully sharing it. Find someone in private to show you how. It's easily possible once you know.

Good luck

closed 2026-09-29 by prusaslicer-bot[bot]