prusa3d/PrusaSlicer #4234: PrusaSlicer no longer opens after attempting to install filament profile

Merlin04, 2020-05-13:
### Version
2.2.0

### Operating system type + version
Ubuntu 19.10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S, latest firmware

### Behavior
I tried adding someone else's filament profile (from version 2.0.0) by putting it into the `filament` folder in `.PrusaSlicer`, but when I tried starting PrusaSlicer I got an error saying the preset contains a bunch of incorrect keys. I tried removing it, but when I start PrusaSlicer, I now get this error:
```
Gtk-Message: 17:33:12.463: Failed to load module "canberra-gtk-module"
The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 147 error_code 2 request_code 152 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```
I've tried deleting my `.PrusaSlicer` folder and re-downloading the AppImage but neither of those have helped.

JohnnyDeer, 2020-05-13:
Hi @Merlin04 , could you please upload "profile" witch you putted into the filament folder? Ideally with most information how you did it, for better issue reproducing. Thank you.

Merlin04, 2020-05-13:
Here's the profile:
https://github.com/eoprede/prusa_profiles/blob/master/MK3/Slic3r/Pretty%20PLA%20V4.0.ini
I copied the INI file into the filament folder, after it didn't work I tried putting it in a few other folders (I think one was presets), which didn't work. I then deleted the INI file and had the issue with PrusaSlicer not starting.

Merlin04, 2020-05-16:
Not sure why but it is working now.

closed 2020-05-16 by Merlin04