prusa3d/PrusaSlicer #4237: The printer maximum acceleration when extruding value limits all other print accelerations

eschulma, 2020-05-13:
Version 2.2.0
Windows 10
Ender 3 Pro, running Marlin 2.0 firmware

Prusa Slicer uses _Machine Limits -> Maximum acceleration when extruding_ to set the initial default acceleration and to cap all other accelerations. If that value is lower than the other accelerations, including the x,y,z, or the ones set in the Print settings, the slicer will add an M204 at the beginning of the file and ignore anything that was set higher. Needless to say this makes the prints much slower.

I am attaching two 3mf files with gcode. Note that the print default acceleration should be 1000, and infill 1500. If I set the _Machine Limits -> Maximum acceleration when extruding_ to 500, I get an M204 500 right before the print begins, which is too low, and it does not change for the infill either.

If I If I set the _Machine Limits -> Maximum acceleration when extruding_ to 1500, I get an M204 S1000 at the start of the print (expect from my settings), and M204 S1500 for the start of the infill, as expected.

This slows this print from 1hr 30 to 1hr 49 minutes, and the effect is worse for longer prints. I am attaching the relevant configuration files.

[PROBLEM config.ini.txt](https://github.com/prusa3d/PrusaSlicer/files/4622967/PROBLEM.config.ini.txt)

[working config.ini.txt](https://github.com/prusa3d/PrusaSlicer/files/4622979/working.config.ini.txt)

rtyr, 2020-05-13:
This is basically a duplicate of  #3443. I agree that these settings needs some rework mainly to respect current Marlin M204 format. (see https://github.com/prusa3d/PrusaSlicer/issues/1089). Your example seems like expected behavior with the current implementation in PS. "Maximum acceleration when extruding" might not have the proper meaning according to Marlin implementation, but still it is a limit.

eschulma, 2020-05-13:
It's related to but not the same as #3443 . That one correctly points out the confusion. But here the setting is silently SLOWING THE PRINT which is more serious, imho. At a minimum, if you set print accelerations that are greater than this limit, there should be a warning similar to the one you get when setting a layer height greater than the extruder limit.

bubnikv, 2021-07-29:
PrusaSlicer 2.4.0-alpha1 will implement M204 PRT the Marlin 2 way for the new Marlin 2 firmware flavor, thus we have split the acceleration while extruding from the acceleration while traveling. Please test with the upcoming PrusaSlicer 2.4.0-alpha1 and let us know whether you are fine with the solution.

closed 2026-09-29 by prusaslicer-bot[bot]