prusa3d/PrusaSlicer #4238: How to capture the color at each layer?

2blane, 2020-05-13:
Is there a setting or way to capture the filament color at each layer and have that passed into the gcode? Or at least have the color be put in the gcode at each color change?

jesusbackflips, 2020-05-14:
Yes check this here out :) 

-----Layer Shift Bar-----
Using Left-Mouse click for M600 Gcode or Right-Mouse click for more Options ;) You need to Drag it first to see the "+"

[image]

2blane, 2020-05-14:
@jesusbackflips Thanks for your reply! So, I can set color changes in the Slicer just fine. The question was how to get those colors in the exported gcode.
For instance, let’s say I have a part that has 4 color changes. I change from red -> yellow -> green -> blue. In the Slicer, the colors are set correctly and the preview shows the 4 colors. Then, when I export to gcode, the color information is lost. The only thing that tells me there is a color change is the M600 command. There is no comment in the code like “now using the yellow color”, “now using blue”, ...
It seems like this information should be passed into the exported gcode for reference. I know the printer wouldn’t use the color information, but it would be a helpful comment.

A few examples of when this could be useful.
1. 3rd Party Visualizer - a visualizer that turns gcode into a 3d model. Without the color information what color would the 3d model be?
2. OctoPrint - a notification could be sent to your phone telling you to change colors to “green”, “purple”, ... or whatever color is next.
3. 3rd Party Analysis - looking at a print to determine how much filament of each color you will need to complete the print.

As you can see, there are many different use cases for adding the color information into the exported gcode. Do you know if this is possible with the current Slicer?

rtyr, 2020-05-14:
You can use "Add custom gcode" option and insert color name and M600 command there. You can probably even use M117 command to display message on the LCD (something like "insert red").

I am not sure if it is possible to do it automatically in some meaningfull way, because PrusaSlicer would use color names like "#FF8000", which is useless for this purpose. But maybe we can include some short custom message, similar to print pause feature :-/. 

[image]

Panayiotis-git, 2020-05-14:
I've prepared the pull request https://github.com/prusa3d/Prusa-Firmware/pull/2380 for the exact same reason!
Then, the slicer could add the message.

2blane, 2020-05-14:
@Panayiotis-git It's been since December? Not looking good to get your pull request in the Firmware or does it usually take that long?

jesusbackflips, 2020-05-14:
Can you maybe use kind a Script for now?
Like Set your M600 Commands + "Red".
And let Script replace the stuff in gcode how you like it :) 
Like put on other line for Pause the Print what ever

closed 2026-09-29 by prusaslicer-bot[bot]