prusa3d/PrusaSlicer #4240: Prusaslicer not working with screenreader on Mac and crashing out

OliverKennett, 2020-05-14:
Hi, I'm a blind guy trying to use Prusaslicer on a 2018 MacBook Air. I'm using the screenreader Voiceover here which reads out elements on the screen in a synthesised voice. There are many accessibility issues with Prusaslicer that I think could be solved fairly easily, like better labelling of buttons etc but the issue I'm facing at the moment is, with Voiceover running, Prusaslicer is incredibly sluggish and when tabbing through items on the screen, it simply crashes. This means I can't use the software at all so a full evaluation of the accessibility is very difficult. I've not put in an issue like this before so please bear with me. I've listed the system info copied from Prusa slicer below and below that I've included the crash report. 

Please help me sort this, I've got a MK3S on the way as an upgrade from my Ender3 and without using Prusaslicer I fear I'm not going to get the best out of my new purchase without some serious tinkering with home brew profiles.

PrusaSlicer
Version:   2.2.0+
Build:     PrusaSlicer-2.2.0+-202003211132

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Catalina Version 10.15.4 (Build 19E287)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version:   2.1 INTEL-14.5.22
Vendor:       Intel Inc.
Renderer:     Intel(R) UHD Graphics 617
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
[... 590 lines cut ...]
__OBJC_RO                         32.2M        1 
__OBJC_RW                         1892K        2 
__TEXT                           289.5M      327 
__UNICODE                          564K        1 
mapped file                       78.0M       23 
shared memory                      640K       16 
===========                     =======  ======= 
TOTAL                              1.8G     2910 
TOTAL, minus reserved VM space     1.4G     2910 

Model: MacBookAir8,1, BootROM 1037.100.362.0.0 (iBridge: 17.16.14281.0.0,0), 2 processors, Dual-Core Intel Core i5, 1.6 GHz, 8 GB, SMC 
Graphics: kHW_IntelUHDGraphics617Item, Intel UHD Graphics 617, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C3531324D3634443450512D3039
Memory Module: BANK 1/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C3531324D3634443450512D3039
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x843), wl0: Feb 28 2020 15:31:19 version 9.30.357.35.32.5.42 FWID 01-9d05982a
Bluetooth: Version 7.0.4f6, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: Apple T2 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 47.4

tommck, 2020-08-25:
This is a long time for no response from the team.  Is anyone working on accessibility problems?

OliverKennett, 2020-08-25:
We’ve exchanged a few emails and I think they were looking into it for the release after the next. Issue is the framework and the problems that the software already has with Mac. From my understanding it is a bit messy getting it from windows, which I assume is the original build, into Mac. Examples include the combo boxes to select printer and material. These are working on windows but become an inaccessible scroll field on Mac.

closed 2026-09-29 by prusaslicer-bot[bot]