prusa3d/PrusaSlicer #4241: Top solid infill settings cannot be changed in Vase mode

mathiasvr, 2020-05-14:
### Version
2.2.0+

### Operating system type + version
macOS Mojave 10.14.6

### Behavior
* _Describe the problem_
In Vase mode (Spiral vase) it is not possible to change the extrusion width or speed of Top solid infill, the textbox is grayed out. However, top solid infill is being used for the top layer of the bottom, before real spiral vase begins.
As a workaround, one can disable vase mode and add top layer, which then allows changing the width and speed of top solid infill, and then re-enable vase mode.

* _Expected Results_
It should be possible to directly change these settings, or the top solid infill should be replaced by solid infill in vase mode.

* _Actual Results_
Settings that are clearly active cannot be changed directly.

_Is this a new feature request?_
No

WarrenSchultz, 2021-01-01:
This isn't a feature request, it's a bug. 
If you turn off vase mode, you can adjust the settings, and then re-enable vase mode, they stick. Clearly an issue.

jlewin, 2021-01-22:
This is frustrating and unexpected. Hard-coding this speed and taking away user control makes little sense.

Please fix.

mihai-stanimir, 2021-03-07:
Wow this has been here for 10 months.

fiveangle, 2021-08-20:
@rtyr please add the "bug" label to this issue as it is clearly a bug

bubnikv, 2021-08-23:
Fixed with 30bca80cc5a5b74f6d6647f940475888ef595389

closed 2021-08-23 by bubnikv