prusa3d/PrusaSlicer #4242: Saving a project does not save support placement(SL1)

eightheads, 2020-05-14:
### Version
Version 2.2.0+

### Operating system type + version
macOS Catalina
Version 10.15.4

### 3D printer brand / version + firmware version (if known)
Setup as Prusa SL1, using Anycubic Photon

### Behavior
* _Describe the problem_
After placing auto and/or custom supports and saving the project, when the project is re-opened the supports are no longer there.

* _Steps needed to reproduce the problem_
Import STL, Place Supports, Save Project, Close Prusaslicer, Open Prusaslicer, Open Project

* _Expected Results_
Saving a project should save any previously set supports and their settings

* _Actual Results_
When opening a previously saved project all previously places supports are gone. No supports are present.

#### Project File (.3MF) where problem occurs
[Body_Legs.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4631056/Body_Legs.3mf.zip)

DrStalker, 2020-07-27:
It would be really helpful to be able to open a project ad adjust the supports without having to start from scratch every time.

eightheads, 2020-07-27:
> It would be really helpful to be able to open a project ad adjust the supports without having to start from scratch every time.

Exactly

bubnikv, 2021-02-19:
User placed support points are being saved. What is the problem?

eightheads, 2021-02-19:
> User placed support points are being saved. What is the problem?

How? I just updated to the latest v2.3.0 and after placing supports and saving the project, when you re-open the project the supports are lost.

FidelCapo, 2021-03-25:
Only custom support points are saved into 3mf project file. Auto generated support points and generated supports are not saved. When you save and reopen project with placed custom support points, you should see them when you open SLA support gizmo. Support need to be generated again after reopening the project. You need also always apply changes in manual support point gizmo before saving.

closed 2021-09-22 by bubnikv