prusa3d/PrusaSlicer #4244: Not possible to get a truly single wall

KlausHans, 2020-05-14:
### Version
2.2.0

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Self made, Marlin 2.0

### Behavior
I have a very weird behavior if i try to print a simple cube with a wall thickness that corresponds to my extrusion width set in Prusa Slicer. In my case i have set an extrusion width of 0.7 mm (0.6mm Nozzle) and the wall of the model is also 0.7mm.
The gcode preview seems fine, but i noticed, that the printhead takes two passes on each layer for the wall. And the printed wall is ca. 1.15mm thick.
I inspected the gcode and noticed, that it really takes two passes on exact the same x/y-coordinates for the same layer.
Here's a snipped of two layers (z=0.8mm and z=1mm):
```
G1 Z0.800
G1 E1.60000 F1800.00000
G1 F1990
G1 X68.726 Y91.887 E1.05448
G1 X68.726 Y72.587 E1.05448
G1 X88.026 Y72.587 E1.05448
G1 X88.026 Y91.797 E1.04957
G1 X88.182 Y92.466 F9000.000
G1 X88.026 Y91.887
G1 F1990
G1 X68.726 Y91.887 E1.05448
G1 X68.726 Y72.587 E1.05448
G1 X88.026 Y72.587 E1.05448
G1 X88.026 Y91.797 E1.04957
G1 X87.447 Y91.732 F9000.000
G1 Z1.000 F9000.000
G1 X88.026 Y91.887
G1 F1881
G1 X68.726 Y91.887 E1.05448
G1 X68.726 Y72.587 E1.05448
G1 X88.026 Y72.587 E1.05448
G1 X88.026 Y91.797 E1.04957
G1 X88.182 Y92.466 F9000.000
G1 X88.026 Y91.887
G1 F1881
G1 X68.726 Y91.887 E1.05448
G1 X68.726 Y72.587 E1.05448
G1 X88.026 Y72.587 E1.05448
G1 X88.026 Y91.797 E1.04957
G1 X87.447 Y91.732 F9000.000
```
For me it looks like a bug.

#### Project File (.3MF) where problem occurs
[07.zip](https://github.com/prusa3d/PrusaSlicer/files/4631236/07.zip)

Thanks!

KlausHans, 2020-05-17:
Am i the only one with this issue? 
This is a major issue in my optinion. I tried the model with Cura, there it works mostly like intended. Only mostly because with Cura i have an extrusion problem (even with no combing or retraction)... I never thought this simple model is such a problem for the two flagship slicers.

HWGeek2020, 2020-05-18:
Try to enable "Detect thin walls".

bubnikv, 2020-12-11:
@KlausHans 
> Try to enable "Detect thin walls",

Does it work for you?

rtyr, 2021-06-08:
No response. Closing.

closed 2021-06-08 by rtyr