prusa3d/PrusaSlicer #4245: PrusaSlicer Bridges Parts when Gap is small

jimvb-ut, 2020-05-15:
### Version 
                        Prusa Slicer Version 2.2.0
### Operating system type + version
                        Windows 10 Pro - v1909
### 3D printer brand / version + firmware version (if known)
                        Original Prusa i3 Mk3S-MMU2s
### Behavior
                        BUG -- Prusa Slicer Completely ignores Perimeters when gap between parts is small.  Slicing Planetary Gear Bearing with small tolerance (to make the Bearing less sloppy) causes bridging between the Planet gears and the Ring gear (see screen prints) - this is in the Slicer and NOT a printer problem.

                       This is even more obvious when slicing the Makers Muse Tolerance Test v3.0 STL file.  The 0.05 mm test plug is completely incorporated into the base. (see screen prints).

                       These problems ARE associated with the Prusa Slicer.  I have tested the same files with Cura (v4.61) and these issues are not present in the sliced file.

                        I fully realize that the Printer may not be able to successfully print these Models, but that is the whole reason to print the Tolerance Test and if I use the Prusa Slicer I will not get a valid result because the smallest gap is solidly attached

#### Project File (.3MF) where problem occurs

[image]

[image]

[image]

[image]

[Makers  Muse Tolerance Test v3 - 3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4631767/Makers.Muse.Tolerance.Test.v3.-.3mf.zip)

neophyl, 2020-05-15:
If you go to Print Settings>Advanced>Slicing in either Advanced or expert mode you will see a setting called 'Slice Gap closing radius'.  By default it is set to 0.0049mm.

This setting is designed to fill in small cracks of the model.  If you read the tooltip it fills in gaps 2x the closing radius.  It is this setting casing the slicing issue.  I added an extra zero to make it 0.00049 and it then slices without a problem.

jimvb-ut, 2020-05-15:
Wow! Thank you for responding so quickly.

So this is a Feature - not a bug.  Thanks for pointing this out to me, I
don't think that I ever noticed this setting before.

This works as you say, and slices correctly when I reduced the 'Gap closing
radius'.  However, for some reason my copy of Slicer seems to think that
the default value is 0.049 instead of 0.0049 as you stated above. Actually
the 0.0049 value seems sufficient for my needs right now.

Anyway this works.  Thanks again.

Jim von Behren

neophyl, 2020-05-15:
Yeah sorry I miss read the value and it was only after posting that I noticed which is why I edited my post.

bubnikv, 2023-06-19:
Solved, closing.

closed 2023-06-19 by bubnikv