prusa3d/PrusaSlicer #4246: [Feature Request] Concentric infill with radii.

James-G4CLF, 2020-05-15:
It would be helpful with low infill percentages if there were a version of the concentric infill with radial connections between the concentric shapes. I am not a mathematician so refrain from suggesting how to define the radii positions but do attach a sketch of how circular infill might look.
[image]

bubnikv, 2020-05-15:
It looks nice on a drawing, but I am not sure how useful it would be for the FDM technology.

1) Concetric circles need to be aligned or nearly aligned to overlap in Z.
2) FDM technology loves continues draws of extrusions, which is non trivial and not always or fully possible to achieve for such a pattern.

73, Vojtech OK1IAK

James-G4CLF, 2020-05-15:
1. Not sure I understand the concentric circle issue. Isn't it solved in "Concentric" infill?

2. That's why I emphasised my lack of mathematical skills. I am aware that for continuity there must be an even number of lines meeting at a node - which is not the case in my sketch (85 nodes, 72 with 3 lines, just 13 with 4). But might there be an approximation which improves the concentric rigidity and does have mostly even nodes? Even if the "radials" are not truly radial? I want strong printed cylinders without totally solid walls.

73 de G4CLF - James

jesusbackflips, 2020-05-17:
What about Waveshape style? Because Nozzle cant hit so hard in Vertical, Horzintal lines :) A Grid but Wavy .)
(view from on Top, so its flat in Vertical Dimension)
[image]

James-G4CLF, 2020-05-18:
I think what I am asking for is analagous to brick ties in a cavity wall. Where I do not want a solid "thin wall" but want ties across the gap:-
[image]

Jaxx005, 2021-02-07:
A radial infill would certainly be useful for internal cylinders, that require mechanical strength/rigidity/support in all directions, about those cylindrical perimeters.  The existing infills don't quite meet this requirement.   

A 'radial' or radiating infill (with 'walls' eminating from a centre) would also be useful in modifier shapes/layer sections, such as for locally reinforcing internal cylinders (e.g. holes for bolts etc).  For that reason, I would prefer this to be a solid-walled infill, like Grid.

The drawing above (James-G4CLF) showing the 'castellated' infill would be useful.  I was thinking perhaps more like a radial or 'polar' _grid_, or spider-web pattern?  The %infill value would govern the number of radials and/or spacing of concentric (cylindrical) infill walls  

The number and length of radials would affect strength and rigidity. So I suppose the %infill value would be best to just set the number of radials (concentric wall at preset spacing)? Else set the 'scale' of grid pattern as a whole? Or perhaps this warrants separate types of 'radial grid'?

This could possibly be adapted to an _adaptive_ infill, perhaps "Adaptive Polar Grid"?  With more radials nearer to surfaces, with % adjusting density of radials?

The current method of achieving 'radial webs' for mechanical strength/rigidity between internal features, is to draw them in CAD. And it could be argued that such mechanical design aspects should be dealt with in CAD, not in a slicer!  But quite often a radial infill of some kind would be sufficient.    :)  

Example:
[image]

mainr, 2021-04-11:
I'm also no Math-magician, but I'd like to suggest four changes to what's being proposed for what I'd generally term 'curve-aware' infill:
1. Support cylindrical infill with either 'radial' or 'axial' alignment (what is shown in this thread until now is 'axial' alignment only).
2. Support all existing forms of infill (i.e., gyroid, triangles, etc.), but apply a transformation (based on the alignment selected from step 1), such that infill dimensions change with radius to maintain a consistent wall thickness.
3. Support radial alignment in 3 dimensions (i.e., for shapes curved in 3D, for instance, spheres, but could also apply against 3D polygons, such as dodecahedron). Think of a football (soccer) ball, with the edges of the pentagons/hexagons defining the walls of the infill.
4. Expand infill options. Currently honeycomb is limited to hexagon. Pentagon & octagon would also be interesting, and shouldn't involve too much additional coding.

Apart from possible mechanical advantages of 'curve-aware' infill, the above changes could create particularly atheistically-pleasing features when left exposed (no walls), such as in [this video](https://www.youtube.com/watch?v=6PVeh43Or-g)

cylindrical - axial aligned:
[image]

cylindrical - radially aligned:
[image]

James-G4CLF, 2026-09-18:
I think that these are useful suggestions and should be considered in upgrades of PrusaSlicer 3