prusa3d/PrusaSlicer #4250: Incorrect slicing

ucacademy, 2020-05-15:
### Version
2.2.0+win64

### Operating system type + version
windows 10 64bit

### 3D printer brand / version + firmware version (if known)
creality ender 5

### Behavior
I have an STL model generated by openSCAD, which has a hole inside it.
In 3d view, the hole is there.
When I try slicing (with the settings in the attached project) the hole becomes closed at a certain height. If i change slightly the size of the inner hole (making it 0.1mm bigger), the problem disappears.
I attached the 3mf project that gives me the problem, and 2 screenshots.

[bug_001.zip](https://github.com/prusa3d/PrusaSlicer/files/4634574/bug_001.zip)
[image]
[image]

rtyr, 2020-05-15:
Repairing the model will fix the issue.

[image]

ucacademy, 2020-05-15:
Thanks for the fast reply :)

Is there any way to spot what is wrong in the model? 
It is generated with openscad with intersection of rotational extrudes, so I can't explain why is should be wrong.. I know it's not a slicer issue at this point, just asking to know if slicer can help me in debugging what is going on with model..

Thanks!

rtyr, 2020-05-15:
There is some brief info, but not graphic representation. But I can see that there is some second (most likely unwanted) shell, which is barely visible.

[image]
[image]

jesusbackflips, 2020-05-15:
Check "3D Paint" (Microsoft "App" Store).
I think you can Fix and also build easy stuff :) And free :D

FidelCapo, 2021-02-01:
I can't test the fix for this model because there is no stl file available, only a 3mf file. However, according to tests of other models, this issue should be corrected in next release.

thepyper, 2021-02-02:
I'm sorry, I don't have this file anymore, I just corrected it at the time and the buggy file is gone..
I thought the 3mf contained the pertinent data, isn't it? Downloading the bug_001.3mf and opening it with
prusaslicer (current version, 2.3.0) shows a model to me.. and slices it correctly too :)

FidelCapo, 2021-02-02:
@thepyper Thanks for answer. Exporting 3mf very often fixes the original model so I cannot use it for tests. But this issue should be fixed.

closed 2021-02-02 by FidelCapo