prusa3d/PrusaSlicer #4251: Request: Warn on import when vertices "coincide"

smurfix, 2020-05-15:
### Version
2.2.0, current Alpha

### Behavior
Models generated by OpenSCAD often are perfectly manifold, yet do not render correctly. I have thrown away too many prints because PrusaSlicer "forgot" crucial details. I am attaching one of these models:
[frame.stl-not-.zip](https://github.com/prusa3d/PrusaSlicer/files/4636031/frame.stl-not-.zip)

Observe the detail in the attached image:
[image]
[image]

Neither Blender nor PrusaSlicer complains that the model is non-manifold.

Fix: open the STL file in Blender, Vertex > Merge > by distance (10 µm or so). After that step, the model gets sliced perfectly.

PrucaSlicer 2.0 didn't have this problem. If this regression cannot be fixed for 2.3, there should at least be a warning. That should be reasonably easy to implement.

FidelCapo, 2021-02-01:
I can't test the fix for this model because fo some reason I cannot unzip this model. However, according to tests of other models, this issue should be corrected in next release.
Closing.

rtyr, 2021-06-08:
I was able to reproduce it with 2.3.1, but it is sliced correctly on current master (attached file is STL renamed to zip).

Fixed with https://github.com/prusa3d/PrusaSlicer/commit/9c5dd358ab23e9351dce0b9c71eaa5ddf92d6f17. Closing.

closed 2021-06-08 by rtyr