prusa3d/PrusaSlicer #4253: Empty layers detected

Chiron2020, 2020-05-16:
### Version
PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
Windows 10 x64 
Version 1809 (OS_Build 17763.1098

### 3D printer brand / version + firmware version (if known)
Anycubic Chiron
Firmware 1.30

### Behavior
Import Wormgear from Thingivers, 
3D View shows as expected, after slicing missing layers with Error message:

Empty layers detected, the output would not be printable.
Object name: wormonly.stl
Print z: 34.850000
This is usually caused by negligibly small extrusions or by a faulty model. Try to repair the model or change its orientation on the bed.

STL file works in Slic3D and cura no probs.

[WormGear.zip](https://github.com/prusa3d/PrusaSlicer/files/4638358/WormGear.zip)
[image]
[image]

rtyr, 2020-05-16:
Reapiring the model will solve the issue.

[image]

Chiron2020, 2020-05-17:
Sorry, and thank you very much. That solved it. It stared wright into my face and I didn't see it.

closed 2020-05-17 by Chiron2020