prusa3d/PrusaSlicer #4255: Debian Buster: Make fails in dep_blosc

m3cinc, 2020-05-16:
### Version
2.3.0alpha0

### Operating system type + version
Debian10 Buster

### Behavior
* _Describe the problem_
make cannot succeed and fails with following message in building dep_blosc
[ 50%] Creating directories for 'dep_blosc'
[ 51%] Performing download step (git clone) for 'dep_blosc'
Cloning into 'dep_blosc'...
Note: checking out 'e63775855294b50820ef44d1b157f4de1cc38d3e'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at e637758 Getting ready for 1.17.0 release
[ 51%] Performing patch step for 'dep_blosc'
HEAD is now at e637758 Getting ready for 1.17.0 release
/home/pi/share/PSSK/deps/blosc-mods.patch:398: trailing whitespace.
include(CMakeFindDependencyMacro)
/home/pi/share/PSSK/deps/blosc-mods.patch:399: trailing whitespace.

/home/pi/share/PSSK/deps/blosc-mods.patch:400: trailing whitespace.
include("${CMAKE_CURRENT_LIST_DIR}/BloscTargets.cmake")
/home/pi/share/PSSK/deps/blosc-mods.patch:401: trailing whitespace.

/home/pi/share/PSSK/deps/blosc-mods.patch:402: trailing whitespace.
function(_blosc_remap_configs from_Cfg to_Cfg)
error: patch failed: CMakeLists.txt:71
error: CMakeLists.txt: patch does not apply
error: patch failed: blosc/CMakeLists.txt:1
error: blosc/CMakeLists.txt: patch does not apply
error: patch failed: cmake/FindLZ4.cmake:5
error: cmake/FindLZ4.cmake: patch does not apply
error: patch failed: cmake/FindSnappy.cmake:3
error: cmake/FindSnappy.cmake: patch does not apply
error: patch failed: cmake/FindZstd.cmake:3
error: cmake/FindZstd.cmake: patch does not apply
make[2]: *** [CMakeFiles/dep_blosc.dir/build.make:120: dep_blosc-prefix/src/dep_blosc-stamp/dep_blosc-patch] Error 1
make[1]: *** [CMakeFiles/Makefile2:289: CMakeFiles/dep_blosc.dir/all] Error 2
make: *** [Makefile:104: all] Error 2

* _Steps needed to reproduce the problem_
follow the linux build recommendation:
cd deps
mkdir build
cmake ..
make

tamasmeszaros, 2020-05-18:
In a clean docker image based on `debian:buster`, dep_blosc target builds fine. Please try to clone the project from scratch, check your git instance and version, etc...

m3cinc, 2020-05-18:
This issue surfaced on a 4GB Raspberry Pi4, not a docker image.
I will attempt a new image as clean as possible.

tamasmeszaros, 2020-05-19:
Raspbian also builds fine for us.

m3cinc, 2020-05-20:
Unfortunately could not build. Would you mind sharing the loaded libraries/ That could be helpful. I also tried with the latest released git version 2.26.2 and stalled exactly at the same point. Thanks in advance.

ayourk, 2020-05-21:
Please check to see if you are missing any of the following:

libcairo2-dev
libexpat1-dev
libglew-dev
libglib2.0-dev
libglib2.0-dev-bin
libglu1-mesa-dev
libglx-dev
libgtk-3-dev
libgtk2.0-dev
libicu-dev
libjpeg-dev
libjpeg-turbo8-dev
libjpeg8-dev
libpng-dev
libsdl1.2-dev
libssl-dev
libtiff-dev
libtiff5-dev
libudev-dev
libx11-dev
libxml2-dev
pkg-config
build-essential install

m3cinc, 2020-05-24:
Raspbian build fine on a fresh install with the dependencies indicated, except for libjpeg8-dev. The 32-bit version run well, but the plate rendition seems out-of-focus?
[image]

ayourk, 2020-05-24:
There are several different versions of the jpeg development libraries.  My Ubuntu Mate install has libjpeg62-dev as an example by default which seems to translate to v1.5.2.  jpeg is a lossless picture compression format.  I could see this being part of the issue in terms of picture quality.

ayourk, 2020-06-04:
Maybe we need direction from the devs as to which version of the JPEG development libraries we should be using.

Zippersk, 2021-01-09:
In my case, this error was caused by cloning the PrusaSlicer repo on windows and that copy it to Linux without changing line endings.

closed 2026-09-29 by prusaslicer-bot[bot]