prusa3d/PrusaSlicer #4259: Building on Linux: undefinied reference to nlopt_get_errmsg

MakerMatrix, 2020-05-17:
### Version
version_2.3.0-alpha0-60-g5c1d736

### Operating system type + version
CentOS7 with devtoolset-7 software collection.

### Behavior
At the "Linking CXX executable prusa-slicer" stage, I get the following error message and the build fails to proceed:

libslic3r/liblibslic3r.a(Arrange.cpp.o): In function `nlopt::opt::get_errmsg() const [clone .isra.65]':
Arrange.cpp:(.text._ZNK5nlopt3opt10get_errmsgEv.isra.65[_ZNK5nlopt3opt7mythrowE12nlopt_result]+0x6): undefined reference to `nlopt_get_errmsg'
libslic3r/liblibslic3r.a(Rotfinder.cpp.o): In function `double libnest2d::opt::NloptOptimizer::optfunc<Slic3r::sla::find_best_rotation(Slic3r::ModelObject const&, float, std::function<void (unsigned int)>, std::function<bool ()>)::{lambda(double, double, double)#1}&, double, double, double>(std::vector<double, std::allocator<double> > const&, std::allocator<double>&, void*)':
Rotfinder.cpp:(.text+0xab4): undefined reference to `nlopt_get_errmsg'
Rotfinder.cpp:(.text+0xad3): undefined reference to `nlopt_get_errmsg'
Rotfinder.cpp:(.text+0xb1c): undefined reference to `nlopt_get_errmsg'
Rotfinder.cpp:(.text+0xb3b): undefined reference to `nlopt_get_errmsg'
libslic3r/liblibslic3r.a(Rotfinder.cpp.o):Rotfinder.cpp:(.text+0x139f): more undefined references to `nlopt_get_errmsg' follow
collect2: error: ld returned 1 exit status

I have the NLopt-devel-2.4.2-2 package installed.  The includes are in /usr/include.  The libs are in /usr/lib64:

bash-4.2$ ls /usr/lib64/libnlopt*
/usr/lib64/libnlopt_cxx.so  /usr/lib64/libnlopt_cxx.so.0  /usr/lib64/libnlopt_cxx.so.0.8.2
bash-4.2$ ls /usr/include/nlopt*
/usr/include/nlopt.f  /usr/include/nlopt.h  /usr/include/nlopt.hpp

Any ideas?

MakerMatrix, 2020-05-18:
Looking in my /usr/include/nlopt.h file shows that in fact the NLopt version (2.4.2?) that my distro installed does not have the prototype for for nlopt_get_errmsg.  So I installed the latest NLopt (2.6.2) from source (in /usr/local), set the NLOPT environment to /usr/local, make a soft link for libnlopt_cxx.so and reran cmake in the main PrusaSlicer build directory.  It then found this new version in /usr/local.  The software then linked.

So, it seems that whatever cmake is doing to check for an adequate version of NLopt is not working in every case.  I guess the cmake/modules/FindNLopt.cmake file needs to be patched somehow.  I'm farily naive about how to make cmake do my bidding or I'd suggest a patch myself.

tamasmeszaros, 2020-05-18:
Thanks for the report. NLopt 2.4.2 from the dev packages of Ubuntu18.04 used to link fine, this should be investigated but we will have to find the cycles for this.

MakerMatrix, 2021-05-06:
> Hey, facing the same issue, is there any way to solve this error?
>
> CMakeFiles/optimizer_node.dir/src/optimizer.cpp.o: In function Optimizer::Func_Prediction()': optimizer.cpp:(.text+0x41b): undefined reference to nlopt_get_errmsg'
> optimizer.cpp:(.text+0x439): undefined reference to nlopt_get_errmsg' optimizer.cpp:(.text+0x47b): undefined reference to nlopt_get_errmsg'
> optimizer.cpp:(.text+0x499): undefined reference to nlopt_get_errmsg' CMakeFiles/optimizer_node.dir/src/optimizer.cpp.o: In function nlopt::opt::mythrow(nlopt_result) const':
> optimizer.cpp:(.text._ZNK5nlopt3opt7mythrowE12nlopt_result[_ZNK5nlopt3opt7mythrowE12nlopt_result]+0x42): undefined reference to nlopt_get_errmsg' CMakeFiles/optimizer_node.dir/src/optimizer.cpp.o:optimizer.cpp:(.text._ZNK5nlopt3opt7mythrowE12nlopt_result[_ZNK5nlopt3opt7mythrowE12nlopt_result]+0x5f): more undefined references to nlopt_get_errmsg' follow

In my case installing nlopt 2.6.2 from source in /usr/local fixed it.  CMake only requires version 2.4.2 but it seems that on some distros, that package does not have the nlopt_get_errmsg function and some pieces of the PrusaSlicer code needs that…

 

HTH,

 

Jarrod

optimizer/CMakeFiles/optimizer_node.dir/build.make:116: recipe for target '/home/pradheep/neobotix_workspace/devel/lib/optimizer/optimizer_node' failed
make[2]: *** [/home/pradheep/neobotix_workspace/devel/lib/optimizer/optimizer_node] Error 1
CMakeFiles/Makefile2:6746: recipe for target 'optimizer/CMakeFiles/optimizer_node.dir/all' failed
make[1]: *** [optimizer/CMakeFiles/optimizer_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

closed 2021-05-06 by MakerMatrix

ashBabu, 2021-05-16:
Hi, same problem for me as well. Intalled NLopt yesterday. The example program given in the documentation runs fine when I compile using ``` g++ nlopt_test.cpp -L/usr/local/lib -lnlopt -lm -o nlopt_test ``` but throws  ```CMakeFiles/main.dir/main.cpp.o: In function `nlopt::opt::get_errmsg() const':/usr/local/include/nlopt.hpp:516: undefined reference to `nlopt_get_errmsg'
collect2: error: ld returned 1 exit status``` when I use the following ```CMakeLists.txt```

```CMake
cmake_minimum_required(VERSION 3.0.0)
project(main VERSION 0.1.0)

include(CTest)
enable_testing()
SET(NLopt_LIBRARY "/usr/local/lib/libnlopt.so")
SET(NLopt_LIBRARIES "/usr/local/lib/libnlopt.so.0" "/usr/local/lib/libnlopt.so.0.11.0")

add_executable(${PROJECT_NAME} main.cpp)

target_include_directories(${PROJECT_NAME} 
    PUBLIC "/usr/local/include", 
    PUBLIC "/usr/local/lib", 
    PUBLIC "/usr/local/include/kdl"
)

target_link_libraries(${PROJECT_NAME} orocos-kdl nlopt m)

include(InstallRequiredSystemLibraries)
set(CPACK_PROJECT_NAME ${PROJECT_NAME})
set(CPACK_PROJECT_VERSION ${PROJECT_VERSION})
include(CPack)
```

Any help?

padhupradheep, 2021-05-16:
Just install it from command line! That should fix it! Before doing that you need to uninstall the pre loaded binaries

ashBabu, 2021-05-16:
Thanks for the quick reply. I had installed ```libnlopt-dev``` using Synaptic Package manager which I think caused the problem.