prusa3d/PrusaSlicer #4260: Issue building dependencies from GIT on Ubuntu 20.04 64-bit

ayourk, 2020-05-17:
### Version
Latest git clone source version_2.3.0-alpha0-60-g5c1d73682 / 5c1d73682d8e347a7dd06c9ae8b1025f3e7a9021

### Operating system type + version
Ubuntu 20.04 64-bit with relevant packages installed:
libcairo2-dev:amd64                             install
libexpat1-dev:amd64                             install
libglew-dev:amd64                               install
libglib2.0-dev:amd64                            install
libglib2.0-dev-bin                              install
libglu1-mesa-dev:amd64                          install
libglx-dev:amd64                                install
libgtk-3-dev:amd64                              install
libgtk2.0-dev:amd64                             install
libicu-dev:amd64                                install
libjpeg-dev:amd64                               install
libjpeg-turbo8-dev:amd64                        install
libjpeg8-dev:amd64                              install
libpng-dev:amd64                                install
libsdl1.2-dev                                   install
libssl-dev:amd64                                install
libtiff-dev:amd64                               install
libtiff5-dev                                    install
libudev-dev:amd64                               install
libx11-dev:amd64                                install
libxml2-dev:amd64                               install
pkg-config                                      install
build-essential                                 install

uname -a:
Linux MyHOST 5.4.0-31-generic #35-Ubuntu SMP Thu May 7 20:20:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2 firmware 3.0.12

### Behavior
I'm trying to build PrusaSlicer on Linux starting with the deps/build subfolder:
user@MyHOST:PrusaSlicer/deps/build>cmake -DCMAKE_BUILD_TYPE=Debug -DDESTDIR=/home/user/custom/deps ..

The above completes without issue.  Then I start the 'make' in that same directory.  Eventually I get:
Scanning dependencies of target stc
Scanning dependencies of target media
[ 88%] Building CXX object libs/stc/CMakeFiles/stc.dir/__/__/__/__/src/stc/stc.cpp.o
Scanning dependencies of target gl
[ 88%] Building CXX object libs/media/CMakeFiles/media.dir/__/__/__/__/src/common/mediactrlcmn.cpp.o
[ 88%] Building CXX object libs/gl/CMakeFiles/gl.dir/__/__/__/__/src/common/glcmn.cpp.o
[ 88%] Building CXX object libs/html/CMakeFiles/html.dir/__/__/__/__/src/html/m_tables.cpp.o
[ 88%] Building CXX object libs/media/CMakeFiles/media.dir/__/__/__/__/src/unix/mediactrl.cpp.o
In file included from /usr/include/gstreamer-0.10/gst/gstelement.h:55,
                 from /usr/include/gstreamer-0.10/gst/gstbin.h:27,
                 from /usr/include/gstreamer-0.10/gst/gst.h:35,
                 from /home/user/custom/slicer/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/unix/mediactrl.cpp:18:
/usr/include/gstreamer-0.10/gst/gstconfig.h:200:11: fatal error: libxml/parser.h: No such file or directory
  200 | # include <libxml/parser.h>
      |           ^~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [libs/media/CMakeFiles/media.dir/build.make:92: libs/media/CMakeFiles/media.dir/__/__/__/__/src/unix/mediactrl.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:1283: libs/media/CMakeFiles/media.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
[ 88%] Building CXX object libs/stc/CMakeFiles/stc.dir/__/__/__/__/src/stc/PlatWX.cpp.o
[ 89%] Building CXX object libs/gl/CMakeFiles/gl.dir/__/__/__/__/src/gtk/glcanvas.cpp.o
[ 89%] Building CXX object libs/html/CMakeFiles/html.dir/__/__/__/__/src/html/styleparams.cpp.o
In file included from /usr/include/gtk-2.0/gtk/gtkobject.h:37,
                 from /usr/include/gtk-2.0/gtk/gtkwidget.h:36,
                 from /usr/include/gtk-2.0/gtk/gtkcontainer.h:35,
                 from /usr/include/gtk-2.0/gtk/gtkbin.h:35,
                 from /usr/include/gtk-2.0/gtk/gtkwindow.h:36,
                 from /usr/include/gtk-2.0/gtk/gtkdialog.h:35,
                 from /usr/include/gtk-2.0/gtk/gtkaboutdialog.h:32,
                 from /usr/include/gtk-2.0/gtk/gtk.h:33,
                 from /home/user/custom/slicer/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/gtk/private/wrapgtk.h:14,
                 from /home/user/custom/slicer/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/glcanvas.cpp:18:
/usr/include/gtk-2.0/gtk/gtktypeutils.h:236:64: warning: ‘GTypeDebugFlags’ is deprecated [-Wdeprecated-declarations]
  236 | void            gtk_type_init   (GTypeDebugFlags    debug_flags);
      |                                                                ^
In file included from /usr/include/glib-2.0/gobject/gobject.h:24,
                 from /usr/include/glib-2.0/gobject/gbinding.h:29,
                 from /usr/include/glib-2.0/glib-object.h:22,
                 from /usr/include/glib-2.0/gio/gioenums.h:28,
                 from /usr/include/glib-2.0/gio/giotypes.h:28,
                 from /usr/include/glib-2.0/gio/gio.h:26,
                 from /usr/include/gtk-2.0/gdk/gdkapplaunchcontext.h:30,
                 from /usr/include/gtk-2.0/gdk/gdk.h:32,
                 from /usr/include/gtk-2.0/gtk/gtk.h:32,
                 from /home/user/custom/slicer/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/gtk/private/wrapgtk.h:14,
                 from /home/user/custom/slicer/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/glcanvas.cpp:18:
/usr/include/glib-2.0/gobject/gtype.h:685:3: note: declared here
  685 | } GTypeDebugFlags GLIB_DEPRECATED_TYPE_IN_2_36;
      |   ^~~~~~~~~~~~~~~
In file included from /usr/include/gtk-2.0/gtk/gtktoolitem.h:31,
                 from /usr/include/gtk-2.0/gtk/gtktoolbutton.h:30,
                 from /usr/include/gtk-2.0/gtk/gtkmenutoolbutton.h:30,
                 from /usr/include/gtk-2.0/gtk/gtk.h:126,
                 from /home/user/custom/slicer/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/gtk/private/wrapgtk.h:14,
                 from /home/user/custom/slicer/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/glcanvas.cpp:18:
/usr/include/gtk-2.0/gtk/gtktooltips.h:73:12: warning: ‘GTimeVal’ is deprecated: Use 'GDateTime' instead [-Wdeprecated-declarations]
   73 |   GTimeVal last_popdown;
      |            ^~~~~~~~~~~~
In file included from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/glib-2.0/gobject/gbinding.h:28,
                 from /usr/include/glib-2.0/glib-object.h:22,
                 from /usr/include/glib-2.0/gio/gioenums.h:28,
                 from /usr/include/glib-2.0/gio/giotypes.h:28,
                 from /usr/include/glib-2.0/gio/gio.h:26,
                 from /usr/include/gtk-2.0/gdk/gdkapplaunchcontext.h:30,
                 from /usr/include/gtk-2.0/gdk/gdk.h:32,
                 from /usr/include/gtk-2.0/gtk/gtk.h:32,
                 from /home/user/custom/slicer/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/gtk/private/wrapgtk.h:14,
                 from /home/user/custom/slicer/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/glcanvas.cpp:18:
/usr/include/glib-2.0/glib/gtypes.h:545:26: note: declared here
  545 | typedef struct _GTimeVal GTimeVal GLIB_DEPRECATED_TYPE_IN_2_62_FOR(GDateTime);
      |                          ^~~~~~~~
[ 89%] Building CXX object libs/html/CMakeFiles/html.dir/__/__/__/__/src/html/winpars.cpp.o
[ 89%] Building CXX object libs/gl/CMakeFiles/gl.dir/__/__/__/__/src/unix/glx11.cpp.o
[ 89%] Building CXX object libs/stc/CMakeFiles/stc.dir/__/__/__/__/src/stc/ScintillaWX.cpp.o
[ 89%] Building CXX object libs/html/CMakeFiles/html.dir/__/__/__/__/src/generic/htmllbox.cpp.o
[ 89%] Linking CXX static library ../../lib/libwx_gtk2u_gl-3.1.a
[ 89%] Built target gl
[ 89%] Linking CXX static library ../../lib/libwx_gtk2u_html-3.1.a
[ 89%] Built target html
[ 89%] Linking CXX static library ../../lib/libwx_gtk2u_stc-3.1.a
[ 89%] Built target stc
make[3]: *** [Makefile:130: all] Error 2
make[2]: *** [CMakeFiles/dep_wxWidgets.dir/build.make:113: dep_wxWidgets-prefix/src/dep_wxWidgets-stamp/dep_wxWidgets-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:134: CMakeFiles/dep_wxWidgets.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
user@MyHOST:PrusaSlicer/deps/build>

I'm pretty sure there is something not making it into the wxWidgets in relation to the fact that I have /usr/include/libxml2/libxml/parser.h instead if /usr/include/lbxml/parser.h.  Somewhere there needs to be a -I/usr/include/libxml2 added to the build process, whether it be from `pkg-config --cflags xml-2.0` or elsewhere.  I'm not sure where to put that correction so the build goes through error free.

MakerMatrix, 2020-05-18:
> : fatal error: libxml/parser.h: No such file or directory

Looks to me like the wxwidgets build needs that to be on your system.  Is there a libxml-devel package you can install?

ayourk, 2020-05-18:
I have it installed as shown in the original post:

libxml2-dev:amd64 install

MakerMatrix, 2020-05-18:
Is libxml/parser.h on your system?  If not, is parser.h on your system somewhere else?  Get apt to tell you which files libxml2-dev installed and where.

ayourk, 2020-05-18:
As stated in the original post, I do have parser.h installed, but the build process is not finding it properly.  My parser.h is located in /usr/include/libxml2/libxml/parser.h which is where the packaging system puts it.  The rest of the build process detects the rest of the dependencies just fine.  "pkg-config --cflags xml-2.0" returns the correct location that the package manager puts them.  The build process doesn't appear to be running that though.

MakerMatrix, 2020-05-18:
The packaging system put it in a different place than Cmake is looking.  That's definitly non-standard.  I think the quick fix is to make soft link like:

ln -s /usr/include/libxml2/libxml /usr/include

The cmake way would be to set the environment variable that cmake (hopefully) obeys to look for libxml includes, to "/usr/include/libxml2"

I have to do this for boost libs on my CentOS7 box.  They get put in a nonstandard place (/usr/lib64/boost169 and /usr/include/boost169).

ayourk, 2020-05-19:
> The cmake way would be to set the environment variable that cmake (hopefully) obeys to look for libxml includes, to "/usr/include/libxml2"
> 
And that's why I opened this issue.  It should be resolved the cmake way and not hackish with symlinks.

MakerMatrix, 2020-05-19:
I'd prefer to do the hack that causes libxml headers appear at a standard place on my system.  But you can probably coax cmake to do the lifting by putting -I/usr/include/libxml2 in CMAKE_CXX_FLAGS and/or CMAKE_C_FLAGS if that's what you prefer.

ayourk, 2020-10-21:
Issue appears to be fixed as of d0ad098661b141b818e592f9369dcf00f0de8cc9.  I haven't done regressions as to which commit fixed this.

closed 2020-10-21 by ayourk